OpenZWave/open-zwave #2285: Barrier value doesn't support exact positions

kpine, 2020-07-02:
Barrier Report::State supports a value that indicates the barrier is stopped at an exact position, with values 0x1 to 0x63. It is optional for devices to support this.

OZW [doesn't handle](https://github.com/OpenZWave/open-zwave/blob/0dba812c2568a59e1cbe9f06c84b5f1adbda3ccb/cpp/src/command_classes/BarrierOperator.cpp#L201) these values. The values for "Closing", "Stopped", "Opening", "Opened" would also collide with exact positions 1%-4%. The application is given the values of the array indices (1-4) instead of the actual values (0xFC-0xFF) for those states.

Fishwaldo, 2020-07-06:
So (without looking) either thats part of a new CC version, or was never implemented in the first Version.

kpine, 2020-07-06:
Probably not implemented in the first place, it's listed as part of V1. I don't actually own a device that supports this, but I was just poking around when reviewing the HA implementation of covers. Figured I'd log this bug in case it was ever needed, as position values 1-4% would collide with the current list indices 1-4 (i.e. a breaking change).

closed 2022-06-06 by kpine