OpenZWave/open-zwave #2286: Remotec ZXT-120

AdamWeglarz, 2020-07-03:
Hi,

There is still issue with Remotec ZXT-120
I know there were discussion here, but it went in very wrong direction which did not lead to solve the problem.
https://github.com/OpenZWave/open-zwave/issues/1070
I have made new patch for driver.cpp to delay removing message if node is ZXT-120
Also I would like to notice that ZXT120 is not maintained anymore so providing support for it on OZW can be done only based on workaround on OZW.
I attach file so if it can pass review I will prepare pull request and if not it can be there that people can compile it by themselves to have support for this device.
[ozw16zxtfix.zip](https://github.com/OpenZWave/open-zwave/files/4869398/ozw16zxtfix.zip)
Adam

AdamWeglarz, 2020-07-03:
Also issue is present on all z-wave sticks I tested it:

- Aeon Z-Stick GEN5
- UZB1

So it is not depended on stick.

Fishwaldo, 2020-07-06:
Sorry - Thats a terrible hack. Basically any message to the device is going to have a 2 second lag. (even messages in the queue to other devices will lag out for 2 seconds). 

These devices are broken. The "fix" is for the vendor to fix their firmware. Closing

closed 2020-07-06 by Fishwaldo

AdamWeglarz, 2020-07-06:
Hello,

I get your point.
But on the other hand all people in the world using openzwave and not having zxt-120 device will not notice the change and these lucky ones like me who have it I think it is better to wait some time (I do not think it is 2 seconds - although I dont know what is usleep implementation in openzwave - if it this 1000x time defined it should be 20 seconds with this parameter hardcoded, but reaction is absolutely without any noticable delay). Like I said this devices will not get fix from supplier as newer revision was provided zxt-600....