OpenZWave/open-zwave #2298: MCOhome FCU-thermostat still not working properly

kostas2911, 2020-07-08:
Hi,

I think that there is an error on manufacturer_specific.xml. id should be 5102 not 3102

Also on mh8fceu0803.xml without below lines:

```
<CommandClass id="67" name="COMMAND_CLASS_THERMOSTAT_SETPOINT" version="2" innif="true" base="1" typeInterpretation="B">
	<Instance index="1" />
	<Value type="decimal" genre="user" instance="1" index="1" label="Heating 1" units="C" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="0" />
	<Value type="decimal" genre="user" instance="1" index="2" label="Cooling 1" units="C" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="0" />
```

i cannot set heat temperature

If this helps, I was using this configuration in zwcfg_0xdbf3b72b.xml in order to work properly.

[mcofcu.txt](https://github.com/OpenZWave/open-zwave/files/4883000/mcofcu.txt)

_Originally posted by @kostas2911 in https://github.com/OpenZWave/open-zwave/issues/2267#issuecomment-654636395_

kostas2911, 2020-07-08:
[image]
Unused 3 should be heating 1

kpine, 2020-07-10:
You could always submit a PR to fix it.

The heat doesn't work because the base and type interpretation in your XML are wrong.
```xml
<CommandClass id="67" name="COMMAND_CLASS_THERMOSTAT_SETPOINT" version="2" innif="true" base="1" typeInterpretation="B">
```
They should be 0 and A. The XML file that was submitted sets the correct values, however if it isn't loaded OZW will use the defaults which are 1 and B.

https://github.com/OpenZWave/open-zwave/blob/752a6ca9c2faf07f872f2a6757a04284be11b410/config/mcohome/mh8fceu0803.xml#L30-L34

kostas2911, 2020-07-10:
What is PR?
I am trying to understand as i am not familiar with.
Just found out that adding 
`<Value type="decimal" genre="user" instance="1" index="1" label="Heating 1" units="C" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="0" />`
makes heating work.

As i understand if xml loaded it will automatically use right values.
But why it is not loading?
Maybe because in manufacturer_specific.xml id is set as 3102 and actual device use 5102?

kpine, 2020-07-10:
> What is PR?

Pull Request is a Github term to upload a change. You could fix the manufacturer xml and correct the ID. If you're not familiar with how to do that then Fishwaldo will probably come along and fix it.

> As i understand if xml loaded it will automatically use right values.
> But why it is not loading?
> Maybe because in manufacturer_specific.xml id is set as 3102 and actual device use 5102?

I think you are correct. If the XML file was loaded correctly, it would find the heating setpoint correctly. You could do that yourself by finding the manufacturer_specific.xml from the config file and change the value.

kostas2911, 2020-07-10:
i run homeassistant and with the new openzwave module it creates a docker as i undestood in which i cannot get access in order to modify files.

On previous zwave module i had access to xmls, thats how i diid the changes.

kpine, 2020-07-10:
Correct, the files are not easily accessible. I know some people have accessed the addon files by using SSH, but I don't know the details. Posting in the HA forums will have better results for that.

kostas2911, 2020-08-01:
Anyone can fix this?

closed 2020-08-15 by kostas2911