OpenZWave/open-zwave #2300: Fibaro FGMS001 ZW5 - no sensor binary report

jghaanstra, 2020-07-10:
I recently started using zwave2mqtt and was migrating my devices from another controller (Homey) to my ZME_UZB1 USB Stick. All is looking good except for one issue which I have been unable to tackle. I noticed that the sensor state (48-1-0) of my Fibaro FGMS001 ZW5 sensor is not updating in zwave2mqtt even though I can see motion detection and cancellation events are coming in as expected (113-1-7).

I initially reported the issue in the zwave2mqtt repo here: https://github.com/OpenZWave/Zwave2Mqtt/issues/621 but after further investigation it looks more related to OZW. After looking into the logs it turns out there are no COMMAND_CLASS_SENSOR_BINARY reports for this sensor registered. Below is the corresponding log. Node 7 is the Fibaro FGMS001. Node 4 is a Neo Coolcam PIR which is working as intended as does send  COMMAND_CLASS_SENSOR_BINARY reports.

```
FIBARO FGMS001 ZW5
2020-07-09T21:07:26.048Z z2m:Zwave zwave node 7: changed: 113-1-7:Home Security:Clear -> Motion Detected at Unknown Location
OpenZWave Detail, Node007,   Received: 0x01, 0x11, 0x00, 0x04, 0x00, 0x07, 0x09, 0x71, 0x05, 0x00, 0x00, 0x00, 0xff, 0x07, 0x08, 0x00, 0xb3, 0x00, 0xd3
OpenZWave Detail, 
OpenZWave Info, Node007, Got a AlarmCmd_Report Message.... 
OpenZWave Info, Node007, Received Notification report (&gt;v1): Type: Home Security (7) Event: Motion Detected at Unknown Location (8) Status: true, Param Length: 0
OpenZWave Detail, Node007, Refreshed Value: old value=3, new value=0, type=byte
OpenZWave Detail, Node007, Changes to this value are not verified
OpenZWave Detail, Node007, Refreshed Value: old value=0, new value=3, type=list
OpenZWave Detail, Node007, Changes to this value are not verified
OpenZWave Detail, Node007, Notification: ValueChanged CC: COMMAND_CLASS_NOTIFICATION Instance: 1 Index: 256
OpenZWave Detail, Node007, Notification: ValueChanged CC: COMMAND_CLASS_NOTIFICATION Instance: 1 Index: 7

2020-07-09T21:07:56.541Z z2m:Zwave zwave node 7: changed: 113-1-256:Previous Event Cleared:0 -> 8
2020-07-09T21:07:56.542Z z2m:Zwave zwave node 7: changed: 113-1-7:Home Security:Motion Detected at Unknown Location -> Clear
OpenZWave Detail, Node007,   Received: 0x01, 0x12, 0x00, 0x04, 0x00, 0x07, 0x0a, 0x71, 0x05, 0x00, 0x00, 0x00, 0xff, 0x07, 0x00, 0x01, 0x08, 0xa7, 0x00, 0xc6
OpenZWave Detail, 
OpenZWave Info, Node007, Got a AlarmCmd_Report Message.... 
OpenZWave Info, Node007, Received Notification report (&gt;v1): Type: Home Security (7) Event: Clear (0) Status: true, Param Length: 1
OpenZWave Detail, Node007, Refreshed Value: old value=0, new value=8, type=byte
OpenZWave Detail, Node007, Changes to this value are not verified
OpenZWave Detail, Node007, Refreshed Value: old value=3, new value=0, type=list
OpenZWave Detail, Node007, Changes to this value are not verified
OpenZWave Detail, Node007, Notification: ValueChanged CC: COMMAND_CLASS_NOTIFICATION Instance: 1 Index: 256
OpenZWave Detail, Node007, Notification: ValueChanged CC: COMMAND_CLASS_NOTIFICATION Instance: 1 Index: 7

NODE 4 NEO COOLCAM PIR
2020-07-09T21:27:13.508Z z2m:Zwave zwave node 4: changed: 113-1-256:Previous Event Cleared:8 -> 0
2020-07-09T21:27:13.510Z z2m:Zwave zwave node 4: changed: 113-1-7:Home Security:Clear -> Motion Detected at Unknown Location
OpenZWave Detail, 
OpenZWave Info, Node004, Got a AlarmCmd_Report Message.... 
OpenZWave Info, Node004, Received Notification report (&gt;v1): Type: Home Security (7) Event: Motion Detected at Unknown Location (8) Status: true, Param Length: 0
OpenZWave Detail, Node004, Refreshed Value: old value=8, new value=0, type=byte
OpenZWave Detail, Node004, Changes to this value are not verified
OpenZWave Detail, Node004, Refreshed Value: old value=0, new value=1, type=list
OpenZWave Detail, Node004, Changes to this value are not verified
OpenZWave Detail, Node004, Notification: ValueChanged CC: COMMAND_CLASS_NOTIFICATION Instance: 1 Index: 256
OpenZWave Detail, Node004, Notification: ValueChanged CC: COMMAND_CLASS_NOTIFICATION Instance: 1 Index: 7
2020-07-09T21:27:13.565Z z2m:Zwave zwave node 4: changed: 48-1-0:Sensor:false -> true
OpenZWave Detail, Node004,   Received: 0x01, 0x0c, 0x00, 0x04, 0x00, 0x04, 0x04, 0x30, 0x03, 0xff, 0x0c, 0xb5, 0x00, 0x82
OpenZWave Detail, 
OpenZWave Info, Node004, Received SensorBinary report: Sensor:12 State=On
OpenZWave Detail, Node004, Refreshed Value: old value=false, new value=true, type=bool
OpenZWave Detail, Node004, Changes to this value are not verified
OpenZWave Detail, Node004, Notification: ValueChanged CC: COMMAND_CLASS_SENSOR_BINARY Instance: 1 Index: 0

2020-07-09T21:27:44.630Z z2m:Zwave zwave node 4: changed: 113-1-256:Previous Event Cleared:0 -> 8
2020-07-09T21:27:44.631Z z2m:Zwave zwave node 4: changed: 113-1-7:Home Security:Motion Detected at Unknown Location -> Clear
OpenZWave Detail, Node004,   Received: 0x01, 0x12, 0x00, 0x04, 0x00, 0x04, 0x0a, 0x71, 0x05, 0x00, 0x00, 0x00, 0xff, 0x07, 0x00, 0x01, 0x08, 0xb5, 0x00, 0xd7
OpenZWave Detail, 
OpenZWave Info, Node004, Got a AlarmCmd_Report Message.... 
OpenZWave Info, Node004, Received Notification report (&gt;v1): Type: Home Security (7) Event: Clear (0) Status: true, Param Length: 1
OpenZWave Detail, Node004, Refreshed Value: old value=0, new value=8, type=byte
OpenZWave Detail, Node004, Changes to this value are not verified
OpenZWave Detail, Node004, Refreshed Value: old value=1, new value=0, type=list
OpenZWave Detail, Node004, Changes to this value are not verified
OpenZWave Detail, Node004, Notification: ValueChanged CC: COMMAND_CLASS_NOTIFICATION Instance: 1 Index: 256
OpenZWave Detail, Node004, Notification: ValueChanged CC: COMMAND_CLASS_NOTIFICATION Instance: 1 Index: 7
2020-07-09T21:27:44.688Z z2m:Zwave zwave node 4: changed: 48-1-0:Sensor:true -> false
OpenZWave Detail, Node004,   Received: 0x01, 0x0c, 0x00, 0x04, 0x00, 0x04, 0x04, 0x30, 0x03, 0x00, 0x0c, 0xb6, 0x00, 0x7e
OpenZWave Detail, 
OpenZWave Info, Node004, Received SensorBinary report: Sensor:12 State=Off
OpenZWave Detail, Node004, Refreshed Value: old value=true, new value=false, type=bool
OpenZWave Detail, Node004, Changes to this value are not verified
OpenZWave Detail, Node004, Notification: ValueChanged CC: COMMAND_CLASS_SENSOR_BINARY Instance: 1 Index: 0
```

I hope this can be looked into and fixed if it's actually a bug. If not, please let me know what I'm doing wrong. If extra information is needed, please let me know as well.

sbik38, 2020-07-19:
I switching actually to Zwave2MQTT too and faced excatly the same issue as you on my FGBS-001. That device is working properly for Motion Detection (in Jeedom so with Openzwave 1.4), but seems that on OZW1.6 (or Zwave2mqtt), the 'sensor' value 48-1-0 is not updated when a motion is present (while the 'home security' 113-1-7 value is changing its states (but in string, not in boolean)

Fishwaldo, 2020-07-20:
>After looking into the logs it turns out there are no COMMAND_CLASS_SENSOR_BINARY reports for this sensor registered.

You should investigate the config options of the device to see if it can be set to send a SENSOR_BINARY (but note - that SENSOR_BINARY is depreciated by the specifications and isn't used on newer devices - So its possible that this device wont support it)

closed 2020-07-20 by Fishwaldo

sbik38, 2020-07-20:
For your information I noutice also the same issue on the Aeotec Multisensotr 6, but this one has the "Sensor Binary" option selectable. By using that parameter the issue dispaear and PIR status updated properly. Fibaro FGMS-001 doesn"t have unfortunately that setting available

petergebruers, 2020-07-20:
If you want the "older" sensor reports... Add your controller to the 4th association group. This group is used for "backwards compatibility" as @Fishwaldo said. The ZW5 by default uses the newer "notification CC".