OpenZWave/open-zwave #2308: Can't change Send interval for ZW096 device

andrerod, 2020-07-12:
Somewhat strange but cant' change the value of interval for a zw096 using ozwadmin. Not sure if it is an openzwave issue (device definition) or a ozwadmin issue though.

andrerod, 2020-07-12:
Was able to set this correctly via MQTT with the sendvalue api. So i think this is somehow a ozwadmin issue.

kpine, 2020-07-13:
https://github.com/OpenZWave/ozw-admin/issues/33

andrerod, 2020-07-14:
@kpine  yeah that seems to be the same issue. 

If it helps for any reader the workaround i did was:
- Open up MQTT Explorer
- Grab the valueidkey from the field definition in the instance node of the device node.
- Use the setvalue command which expects two fields in the payload: the valueidkey and the new value.
- Observe that openzwave replies with an ack / success response and prosper.

andrerod, 2020-07-14:
Closing this here and this is being tracked, correctly, in ozw-admin.

closed 2020-07-14 by andrerod