OpenZWave/open-zwave #2314: Setting half-degree Fahrenheit setpoints for CT101 thermostat works, but getting the setpoint only returns integers

melyux, 2020-07-17:
I have the Radio Thermostat CT101. OZW supports *setting* setpoints (COMMAND_CLASS_THERMOSTAT_SETPOINT) with half-degree precision like 74.5°F, and it works and is reflected on the physical thermostat screen. However, OZW apparently does not support *getting* the thermostat setpoint with half-degree precision.

```
2020-07-17 01:56:11.200 Info, Node023, Value::Set - COMMAND_CLASS_THERMOSTAT_SETPOINT - Cooling 1 - 2 - 1 - 75.5
2020-07-17 01:56:11.605 Info, Node023, Received thermostat setpoint report: Setpoint Cooling 1 = 75F
```

kpine, 2020-07-18:
I have a CT32. If you post the full logs you'll might see the thermostat only reports in whole digits (we can verify if you do), so there isn't much OZW can do about that. At least for the CT32, the display shows the setpoint correctly in 0.5 increments (and the current temperature). But, if you manually increment the values up or down, they only adjust by 1 degree, e.g. 71.5 -> 72.5, or 73 -> 74. You cannot actually adjust in 0.5 increments.

Here's my Set:
```
[20200717 22:06:06.555 PDT] [ozw.library] [info]: Info - Node: 5 Value::Set - COMMAND_CLASS_THERMOSTAT_SETPOINT - Heating 1 - 1 - 1 - 71.5
[20200717 22:06:06.555 PDT] [ozw.library] [debug]: Detail - Node: 5 Queuing (Send) MultiChannel Encapsulated (instance=1): ThermostatSetpointCmd_Set (Node=5): 0x01, 0x11, 0x00, 0x13, 0x05, 0x0a, 0x60, 0x0d, 0x01, 0x01, 0x43, 0x01, 0x01, 0x2a, 0x02, 0xcb, 0x25, 0x3a, 0x20
```

Just to verify the Set is correct, here's the payload:
`0x43, 0x01, 0x01, 0x2a, 0x02, 0xcb`
0x43 = Command Class = COMMAND_CLASS_THERMOSTAT_SETPOINT
0x01 = Command = THERMOSTAT_SETPOINT_SET
0x01 = Setpoint Type = Heating
0x2a = 1 | 01 | 010 = Precision | Scale | Size = 1 decimal place | Fahrenheit | Size of Value (temperature) = 2 Bytes
0x02 = Value Byte 1 = MSB temperature
0xcb = Value Byte 2 = LSB temperature

Then 0x02cb is 715, which is the scaled temp (71.5) in F. Looks good.

Here's the Report for the following Get:
```
[20200717 22:06:08.276 PDT] [ozw.library] [debug]: Detail - Node: 5   Received: 0x01, 0x11, 0x00, 0x04, 0x00, 0x05, 0x09, 0x60, 0x0d, 0x01, 0x01, 0x43, 0x03, 0x01, 0x09, 0x47, 0xca, 0x00, 0x4e
[20200717 22:06:08.276 PDT] [ozw.library] [info]: Info - Node: 5 Response RTT 446 Average Response RTT 326
[20200717 22:06:08.276 PDT] [ozw.library] [info]: Info - Node: 5 Received a MultiChannelEncap from node 5, endpoint 1 for Command Class COMMAND_CLASS_THERMOSTAT_SETPOINT
[20200717 22:06:08.276 PDT] [ozw.library] [debug]: Detail - Node: 5 Refreshed Value: old value=70, new value=71, type=decimal
[20200717 22:06:08.276 PDT] [ozw.library] [debug]: Detail - Node: 5 Changes to this value are not verified
[20200717 22:06:08.277 PDT] [ozw.library] [info]: Info - Node: 5 Received thermostat setpoint report: Setpoint Heating 1 = 71F
```
The Report payload is:
`0x43, 0x03, 0x01, 0x09, 0x47`
0x43 = Command Class = COMMAND_CLASS_THERMOSTAT_SETPOINT
0x03 = Command = THERMOSTAT_SETPOINT_REPORT
0x01 = Setpoint Type = Heating
0x09 = 000 | 01 | 001 = Precision | Scale | Size = 0 decimal places | Fahrenheit | Size of Value (temperature) = 1 Byte
0x47 = Value Byte 1

0x47 = 71, which is the scaled temp (71) in F.

Fishwaldo, 2020-07-20:
@kpine is correct. The device is not sending half degree updates, so we can't do anything about that.

closed 2020-07-20 by Fishwaldo

melyux, 2020-07-20:
@kpine Thanks for investigating this. That's too bad, because setting the half-degree targets works accordingly as well. It's weird that they would make the sets decimal but the gets integer-only for the same thing. I didn't know whether this was a thermostat issue or OZW parsing. But then again, the Radio Thermostat Company seems to have a tenuous grasp on how to do Z-Wave in the first place.