OpenZWave/open-zwave #2324: Honeywell T6 Pro thermostat reports Air Temperature in Celsius until set point is changed

alterbridge86, 2020-07-24:
I recently migrated from the integrated HA OpenZWave distribution to this qt-openzwave container. Everything is running great, I'd say even better than the version integrated into HA, with the exception of this one odd issue I'm seeing with my thermostats which did not occur on OZW 1.4 in HA.

Upon startup of OZW, the thermostat reports Air Temperature in Celsius despite being configured to Fahrenheit locally. Cool and Heat setpoints report in F as expected. On top of this, the Air Temperature value only sends an update sporadically, with no discernable interval. I've seen comments elsewhere (in a SmartThings device handler I think) that this device requires to be polled (ugh...), so for kicks I even tried enabling polling on the Air Temperature value, but it continued to report in Celsius. 

I captured a clean ozwdaemon.log of startup after I removed the ozwcache file, so it should show full device interrogation:

[ozwdaemon.log](https://github.com/OpenZWave/qt-openzwave/files/4854180/ozwdaemon.log)

Nodes 3, 4, and 5 are all T6 Pro thermostats.

The only thing I'm seeing that is _slightly_ suspicious is it appears to choose Celsius as the default scale:

```
[20200630 17:20:00.580 UTC] [ozw.library] [info]: Info - Node: 3 Sending (Send) message (Callback ID=0x66, Expected Reply=0x04) - SensorMultiLevelCmd_SupportedGetScale (Node=3): 0x01, 0x0a, 0x00, 0x13, 0x03, 0x03, 0x31, 0x03, 0x01, 0x25, 0x66, 0x96 
[20200630 17:20:00.624 UTC] [ozw.library] [info]: Info - Node: 3 Received SensorMultiLevel supported Scale report from node 3 for Sensor Air Temperature: C (0) 
[20200630 17:20:00.625 UTC] [ozw.library] [info]: Info - Node: 3 Received SensorMultiLevel supported Scale report from node 3 for Sensor Air Temperature: F (1) 
[20200630 17:20:00.626 UTC] [ozw.library] [info]: Info - Node: 3 Setting SensorMultiLevel Default Scale to: C (0) 
```
and then it reports the current temp in C a little later on:

```
[20200630 17:20:25.799 UTC] [ozw.library] [debug]: Detail - Node: 3 Queuing (Send) SensorMultilevelCmd_Get (Node=3): 0x01, 0x0b, 0x00, 0x13, 0x03, 0x04, 0x31, 0x04, 0x01, 0x00, 0x25, 0xa3, 0x52 
[20200630 17:20:25.805 UTC] [ozw.library] [info]: Info - Node: 3 Sending (Send) message (Callback ID=0xa3, Expected Reply=0x04) - SensorMultilevelCmd_Get (Node=3): 0x01, 0x0b, 0x00, 0x13, 0x03, 0x04, 0x31, 0x04, 0x01, 0x00, 0x25, 0xa3, 0x52 
[20200630 17:20:25.849 UTC] [ozw.library] [debug]: Detail - Node: 3   Received: 0x01, 0x0c, 0x00, 0x04, 0x00, 0x03, 0x06, 0x31, 0x05, 0x01, 0x22, 0x00, 0xdf, 0x3a 
[20200630 17:20:25.850 UTC] [ozw.library] [info]: Info - Node: 3 Response RTT 43 Average Response RTT 42 
[20200630 17:20:25.850 UTC] [ozw.library] [info]: Info - Node: 3 Received SensorMultiLevel report from node 3, instance 1, Air Temperature: value=22.3C 
```

Reason I think this is only _slightly_ suspicious is that my Aeotec Multisensor (Node 7) initializes pretty much identically, setting default scale to C, however it reports temperature in F from the get-go...so this may or may not have any relevance to the issue.

Last bit of info I have to provide - if I go to the thermostat and manually adjust the setpoint, the air temperature will start reporting in F. It's still reported sporadically, but the next report that comes in after adjusting a setpoint manually is in F and Air Temperature remains in F for as long as OZW remains up. As soon as services are restarted again, the air temperature reverts to C and is stuck there until setpoint is adjusted again.

Please let me know if I can provide any further information to help troubleshoot.

Fishwaldo, 2020-07-24:
I've pushed a potential fix. Should be updated when the next ozwdaemon container is pushed out.

alterbridge86, 2020-07-27:
Good news! with a little bad news. But first the good news - on startup I am seeing Fahrenheit values reported off the bat. Logging to confirm:

```
2020-07-27 17:04:36.976866902  [20200727 17:04:36.976 UTC] [ozw.library] [info]: Info - Node: 14 Sending (Send) message (Callback ID=0x45, Expected Reply=0x04) - SensorMultilevelCmd_Get (Node=14): 0x01, 0x0b, 0x00, 0x13, 0x0e, 0x04, 0x31, 0x04, 0x01, 0x08, 0x25, 0x45, 0xb1 
2020-07-27 17:04:36.977350782  [20200727 17:04:36.976 UTC] [ozw.mqtt.publisher] [debug]: Publishing Event valueChanged: 249348116 
2020-07-27 17:04:36.977588269  [20200727 17:04:36.977 UTC] [ozw.library] [info]: Info - Node: 14 Encrypted Flag is 0 
2020-07-27 17:04:36.977921792  [20200727 17:04:36.977 UTC] [ozw.notifications] [debug]: Notification pvt_valueChanged:  281475226058769 Thread:  0x73c70010 
2020-07-27 17:04:36.985786667  [20200727 17:04:36.985 UTC] [ozw.mqtt.publisher] [debug]: Publishing Event valueChanged: 281475226058769 
2020-07-27 17:04:36.988234187  [20200727 17:04:36.986 UTC] [ozw.library] [debug]: Detail - Node: 14   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8 
2020-07-27 17:04:36.989005446  [20200727 17:04:36.988 UTC] [ozw.library] [debug]: Detail - Node: 14   ZW_SEND_DATA delivered to Z-Wave stack 
2020-07-27 17:04:37.007176727  [20200727 17:04:37.006 UTC] [ozw.library] [debug]: Detail - Node: 14   Received: 0x01, 0x07, 0x00, 0x13, 0x45, 0x00, 0x00, 0x03, 0xad 
2020-07-27 17:04:37.007862418  [20200727 17:04:37.007 UTC] [ozw.library] [debug]: Detail - Node: 14   ZW_SEND_DATA Request with callback ID 0x45 received (expected 0x45) 
2020-07-27 17:04:37.008367494  [20200727 17:04:37.008 UTC] [ozw.library] [info]: Info - Node: 14 Request RTT 30 Average Request RTT 69 
2020-07-27 17:04:37.008795335  [20200727 17:04:37.008 UTC] [ozw.library] [debug]: Detail - Node: 14   Expected callbackId was received 
2020-07-27 17:04:37.041270611  [20200727 17:04:37.040 UTC] [ozw.library] [debug]: Detail - Node: 14   Received: 0x01, 0x0c, 0x00, 0x04, 0x00, 0x0e, 0x06, 0x31, 0x05, 0x01, 0x2a, 0x03, 0x16, 0xf5 
2020-07-27 17:04:37.041911253  [20200727 17:04:37.041 UTC] [ozw.library] [info]: Info - Node: 14 Response RTT 64 Average Response RTT 101 
2020-07-27 17:04:37.042453983  [20200727 17:04:37.042 UTC] [ozw.library] [info]: Info - Node: 14 Received SensorMultiLevel report from node 14, instance 1, Air Temperature: value=79.0F 

```
[ozwdaemon_fahrenheit-unit.log](https://github.com/OpenZWave/open-zwave/files/4983345/ozwdaemon_fahrenheit-unit.log)

Now for the bad news. The "Air Temperature Units" are defaulting to Celsius, despite the thermostat being set (on the device itself) to report in Fahrenheit. Only after I went into the ozwcache and updated the Air Temperature Units to vindex=1 did I get the Fahrenheit value reported correctly. Here is logging from when I first started up, prior to changing the Air Temperature Units:

```
2020-07-27 16:48:02.622219343  [20200727 16:48:02.621 UTC] [ozw.library] [info]: Info - Node: 14 Sending (Send) message (Callback ID=0x45, Expected Reply=0x04) - SensorMultilevelCmd_Get (Node=14): 0x01, 0x0b, 0x00, 0x13, 0x0e, 0x04, 0x31, 0x04, 0x01, 0x00, 0x25, 0x45, 0xb9 
2020-07-27 16:48:02.623198716  [20200727 16:48:02.622 UTC] [ozw.library] [info]: Info - Node: 14 Encrypted Flag is 0 
2020-07-27 16:48:02.631013698  [20200727 16:48:02.630 UTC] [ozw.mqtt.publisher] [debug]: Publishing Event valueChanged: 281475226058769 
2020-07-27 16:48:02.632545800  [20200727 16:48:02.632 UTC] [ozw.library] [debug]: Detail - Node: 14   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8 
2020-07-27 16:48:02.633587565  [20200727 16:48:02.633 UTC] [ozw.library] [debug]: Detail - Node: 14   ZW_SEND_DATA delivered to Z-Wave stack 
2020-07-27 16:48:02.652647338  [20200727 16:48:02.652 UTC] [ozw.library] [debug]: Detail - Node: 14   Received: 0x01, 0x07, 0x00, 0x13, 0x45, 0x00, 0x00, 0x03, 0xad 
2020-07-27 16:48:02.653652542  [20200727 16:48:02.653 UTC] [ozw.library] [debug]: Detail - Node: 14   ZW_SEND_DATA Request with callback ID 0x45 received (expected 0x45) 
2020-07-27 16:48:02.654404219  [20200727 16:48:02.654 UTC] [ozw.library] [info]: Info - Node: 14 Request RTT 30 Average Request RTT 70 
2020-07-27 16:48:02.655131158  [20200727 16:48:02.654 UTC] [ozw.library] [debug]: Detail - Node: 14   Expected callbackId was received 
2020-07-27 16:48:02.679096856  [20200727 16:48:02.678 UTC] [ozw.library] [debug]: Detail - Node: 14   Received: 0x01, 0x0c, 0x00, 0x04, 0x00, 0x0e, 0x06, 0x31, 0x05, 0x01, 0x22, 0x01, 0x09, 0xe0 
2020-07-27 16:48:02.680070812  [20200727 16:48:02.679 UTC] [ozw.library] [info]: Info - Node: 14 Response RTT 55 Average Response RTT 96 
2020-07-27 16:48:02.680872955  [20200727 16:48:02.680 UTC] [ozw.library] [info]: Info - Node: 14 Received SensorMultiLevel report from node 14, instance 1, Air Temperature: value=26.5C 
```

From your analysis before I see the request is asking for Celsius (0x31, 0x04, 0x01, **0x00**), which isn't surprising since that's what the config was telling it to do. 

Is it possible the "Air Temperature Units" value is being mis-interpreted when the device is queried? I've noticed that my other Aeotec multisensor (Node 7 in above log files) also shows Celsius for the air temp units, and on boot-up the SensorMultilevelCmd_Get is requesting the value in Celsius as well. Difference here is I have this sensor updating fairly frequently so it is overwritten with a Fahrenheit value before I notice anything is wrong.

If it helps, I just deleted my ozwcache and restarted the daemon. Here is logging to contain all the device querying and the new ozwcache file:

[ozwdaemon_freshstart.log](https://github.com/OpenZWave/open-zwave/files/4983405/ozwdaemon_freshstart.log)
[ozwcache_0xdf2edbb4.txt](https://github.com/OpenZWave/open-zwave/files/4983406/ozwcache_0xdf2edbb4.txt)

joshtbernstein, 2020-08-04:
The issue I was setting seems to be resolved with https://github.com/OpenZWave/open-zwave/commit/21221b69b85f9e67921297e22812f6c8ad8adc2e. Also I do see that Air Temperature Units is set to C as @alterbridge86 noted, but for me the value is also reported in C.

```OpenZWave/1/node/12/instance/1/,{    "Instance": 1,    "TimeStamp": 1596578733}
OpenZWave/1/node/12/instance/1/commandclass/49/,{    "Instance": 1,    "CommandClassId": 49,    "CommandClass": "COMMAND_CLASS_SENSOR_MULTILEVEL",    "CommandClassVersion": 5,    "TimeStamp": 1596578733}
OpenZWave/1/node/12/instance/1/commandclass/49/value/281475183034386/,{    "Label": "Air Temperature",    "Value": 23.5,    "Units": "C",    "ValueSet": true,    "ValuePolled": false,    "ChangeVerified": false,    "Min": 0,    "Max": 0,    "Type": "Decimal",    "Instance": 1,    "CommandClass": "COMMAND_CLASS_SENSOR_MULTILEVEL",    "Index": 1,    "Node": 12,    "Genre": "User",    "Help": "Air Temperature Sensor Value",    "ValueIDKey": 281475183034386,    "ReadOnly": true,    "WriteOnly": false,    "Event": "valueChanged",    "TimeStamp": 1596578751}
OpenZWave/1/node/12/instance/1/commandclass/49/value/1407375089877010/,{    "Label": "Humidity",    "Value": 67.0,    "Units": "%",    "ValueSet": true,    "ValuePolled": false,    "ChangeVerified": false,    "Min": 0,    "Max": 0,    "Type": "Decimal",    "Instance": 1,    "CommandClass": "COMMAND_CLASS_SENSOR_MULTILEVEL",    "Index": 5,    "Node": 12,    "Genre": "User",    "Help": "Humidity Sensor Value",    "ValueIDKey": 1407375089877010,    "ReadOnly": true,    "WriteOnly": false,    "Event": "valueRefreshed",    "TimeStamp": 1596578751}
OpenZWave/1/node/12/instance/1/commandclass/49/value/72057594252640276/,{    "Label": "Air Temperature Units",    "Value": {        "List": [            {                "Value": 0,                "Label": "Celsius"            },            {                "Value": 1,                "Label": "Fahrenheit"            }        ],        "Selected": "Celsius",        "Selected_id": 0    },    "Units": "",    "ValueSet": false,    "ValuePolled": false,    "ChangeVerified": false,    "Min": 0,    "Max": 0,    "Type": "List",    "Instance": 1,    "CommandClass": "COMMAND_CLASS_SENSOR_MULTILEVEL",    "Index": 256,    "Node": 12,    "Genre": "System",    "Help": "Air Temperature Sensor Available Units",    "ValueIDKey": 72057594252640276,    "ReadOnly": false,    "WriteOnly": false,    "Event": "valueAdded",    "TimeStamp": 1596578733}
OpenZWave/1/node/12/instance/1/commandclass/49/value/73183494159482900/,{    "Label": "Humidity Units",    "Value": {        "List": [            {                "Value": 0,                "Label": "Percent"            }        ],        "Selected": "Percent",        "Selected_id": 0    },    "Units": "",    "ValueSet": false,    "ValuePolled": false,    "ChangeVerified": false,    "Min": 0,    "Max": 0,    "Type": "List",    "Instance": 1,    "CommandClass": "COMMAND_CLASS_SENSOR_MULTILEVEL",    "Index": 260,    "Node": 12,    "Genre": "System",    "Help": "Humidity Sensor Available Units",    "ValueIDKey": 73183494159482900,    "ReadOnly": false,    "WriteOnly": false,    "Event": "valueAdded",    "TimeStamp": 1596578733}
OpenZWave/1/node/12/instance/1/commandclass/64/,{    "Instance": 1,    "CommandClassId": 64,    "CommandClass": "COMMAND_CLASS_THERMOSTAT_MODE",    "CommandClassVersion": 3,    "TimeStamp": 1596578733}
OpenZWave/1/node/12/instance/1/commandclass/64/value/206569492/,{    "Label": "Mode",    "Value": {        "List": [            {                "Value": 0,                "Label": "Off"            },            {                "Value": 1,                "Label": "Heat"            },            {                "Value": 2,                "Label": "Cool"            },            {                "Value": 3,                "Label": "Auto"            },            {                "Value": 11,                "Label": "Heat Econ"            },            {                "Value": 12,                "Label": "Cool Econ"            }        ],        "Selected": "Auto",        "Selected_id": 3    },    "Units": "",    "ValueSet": true,    "ValuePolled": false,    "ChangeVerified": false,    "Min": 0,    "Max": 0,    "Type": "List",    "Instance": 1,    "CommandClass": "COMMAND_CLASS_THERMOSTAT_MODE",    "Index": 0,    "Node": 12,    "Genre": "User",    "Help": "Set the Thermostat Mode",    "ValueIDKey": 206569492,    "ReadOnly": false,    "WriteOnly": false,    "Event": "valueChanged",    "TimeStamp": 1596578751}
OpenZWave/1/node/12/instance/1/commandclass/66/,{    "Instance": 1,    "CommandClassId": 66,    "CommandClass": "COMMAND_CLASS_THERMOSTAT_OPERATING_STATE",    "CommandClassVersion": 1,    "TimeStamp": 1596578733}
OpenZWave/1/node/12/instance/1/commandclass/66/value/206602263/,{    "Label": "Operating State",    "Value": "Idle",    "Units": "",    "ValueSet": true,    "ValuePolled": false,    "ChangeVerified": false,    "Min": 0,    "Max": 0,    "Type": "String",    "Instance": 1,    "CommandClass": "COMMAND_CLASS_THERMOSTAT_OPERATING_STATE",    "Index": 0,    "Node": 12,    "Genre": "User",    "Help": "Set the Thermostat Operating State",    "ValueIDKey": 206602263,    "ReadOnly": true,    "WriteOnly": false,    "Event": "valueChanged",    "TimeStamp": 1596578751}
[... cut ...]
OpenZWave/1/node/12/instance/1/commandclass/67/value/2251800020303890/,{    "Label": "Dry Air",    "Value": 0.0,    "Units": "C",    "ValueSet": false,    "ValuePolled": false,    "ChangeVerified": false,    "Min": 0,    "Max": 0,    "Type": "Decimal",    "Instance": 1,    "CommandClass": "COMMAND_CLASS_THERMOSTAT_SETPOINT",    "Index": 8,    "Node": 12,    "Genre": "User",    "Help": "Set the Thermostat Setpoint Dry Air",    "ValueIDKey": 2251800020303890,    "ReadOnly": false,    "WriteOnly": false,    "Event": "valueAdded",    "TimeStamp": 1596578733}```

Fishwaldo, 2020-08-05:
>Now for the bad news. The "Air Temperature Units" are defaulting to Celsius, despite the thermostat being set (on the device itself) to report in Fahrenheit. 

Unfortunately there is no way to get the "scale" that is used on a device as its only transmitted as part of the actual temp reporting. That value is actually used only for specifying what unit we want in response to a GET and does not represent anything on the device itself. 

So if you want to report F, then you need to ensure when you app starts, its always setting the units to F.

alterbridge86, 2020-08-05:
Ok that makes sense. So my final question - how can the app define a "default" scale for OZW? I'm using the qt-openzwave container with Home Assistant, I looked through all of the MQTT commands but don't see anything jumping out at me to define scale.