OpenZWave/open-zwave #2325: Fibaro FGD-212 Dimmer 2 not working (correctly)

alex3305, 2020-07-24:
## Environment

Recently I have migrated my Home Assistant environment to use [OpenZWave Home Assistant Add-On](https://github.com/home-assistant/hassio-addons/tree/master/zwave) (instead of built-in Z-Wave) which in turn is based on [OpenZWave Admin](https://github.com/OpenZWave/ozw-admin). 

All my devices seem to work fine and show up correctly, except my Fibaro FGD-212 Dimmer 2. It's acting very strange and to be honest I don't know what to do with it anymore.

## The problem

* Turning on an instance on the dimmer does work, but only after a delay of about 5 to 10 seconds. The lights will light up really, really dim. Changing brightness has no effect and I cannot turn it off remotely. The physical switch I've connected to the dimmer works and I can turn it off with that
* Sometimes, after changing the brightness more than once it seems, the dimmer will go to the set brightness when flipping the physical switch. Flipping it again turns it off
* Turning the light on with the physical switch results in the light turning on at 20% brightness. Which I cannot change. 
* Double tap seems to work correctly and I can change the brightness once after that
* Refreshing the Node List results in the Node being displayed twice in the Node List
* Reverting to the Home Assistant Z-Wave component solves the issues above

> _**Note** Communication from Home Assistant seems to work fine, thus I'm opening this issue in this repo and not in one of the Home Assistant repo's_

## Logfiles

I have attached two logfiles to a [gist](https://gist.github.com/alex3305/1473cdc32250c326bd38ba145f030f2f) since they were a little large for posting here. The first log file is only with logging for the Node with this issue (Node 8), while the second should be a complete log file from the OpenZWave Docker container.

Fishwaldo, 2020-07-24:
Most Likely Related to this issue: https://github.com/home-assistant/core/issues/38068#issuecomment-663368821

When that fix lands in HA, Please retest

closed 2020-07-24 by Fishwaldo

alex3305, 2020-07-25:
@Fishwaldo Thanks for the feedback. 

I don't know if it is related, but just wanted to let you know that this behaviour also appears when I try to control the light from OZW Admin. I also noticed that the transition times were indeed quite high, but AFAIK I was unable to adjust those values within OZW Admin.

alex3305, 2020-07-27:
I have some additional insights into this issue. Today I have tested OZW Admin without Home Assistant. Thus switching my device from OZW Admin and not with Home Assistant.

I found that the default *Dimming Duration* was quite high at 255. Which seems the max value, as indicated in home-assistant/core#38068. However I could easily bring that back down to something like 1. But refreshing the node reset the *Dimming Duration* to a ridiculous value of **15300**:

[image]

However the light still responds instantly whenever I change the instance level. Perhaps this is a bug on its own?