OpenZWave/open-zwave #2340: After service restart, door locks do not update Alarm Level or Alarm Type

alterbridge86, 2020-08-03:
I've noticed recently that after OZW is restarted, door locks no longer update Alarm Type or Alarm Level values. I have two different deadbolts (a Yale and a Kwikset), with both experiencing the same issue. 

To workaround this, I delete the ozwcache and restart the service. Alarm Type and Alarm Level work again on both locks until the service restarts again (generally every few days due to an OZW container update). 

Here is logging from a fresh startup after deleting the ozwcache file. I toggled the lock state on both Node 5 (Yale) and Node 6 (Kwikset) and the Alarm Type and Alarm Level values updated accordingly.

[current_freshstart.log](https://github.com/OpenZWave/open-zwave/files/5016513/current_freshstart.log)

After this was captured, I shutdown OZW container and started up again. Performed the same test toggling lock state on Node 5 and Node 6, however this time the Alarm Type and Alarm Level did not change from their cached state.

[current_servicerestart.log](https://github.com/OpenZWave/open-zwave/files/5016517/current_servicerestart.log)

Please let me know if you need any additional logging or test cases for further diagnosis.

smartrobot, 2020-08-07:
I am also currently having the same issues with a Yale smart lock.
Both Alarm values are not working as well as the user code that was used

I can provide logging and testing if need be

majorsl, 2020-09-12:
I am seeing this with a Kwikset Smartcode 888 as well. I'd be happy to help however I can.

hoopsta1423, 2020-09-16:
Also seeing this issue with my Yale YRD226 lock, happy to provide logs if needed

bakinbits, 2020-09-23:
Ditto, kwikset 914 here.  happy to help.

pcpete1234, 2020-10-29:
Having the same issue with Kwikset 914.

Running on Windows 10 pc with VirtualBox VM.....cant find that ozwcache file...

Using openzwave addon 0.52 and openzwave beta integration ( OZWDaemon 0.1.0 (OpenZWave 1.6.1210) )

How does one delete this cache file?

alterbridge86, 2020-10-29:
> Having the same issue with Kwikset 914.
> 
> Running on Windows 10 pc with VirtualBox VM.....cant find that ozwcache file...
> 
> Using openzwave addon 0.52 and openzwave beta integration ( OZWDaemon 0.1.0 (OpenZWave 1.6.1210) )
> 
> How does one delete this cache file?

The ozwcache file is in the data directory for ozwdaemon. I don't use the addon but if it is like other addons the data directory should be in your /config directory (probably /config/ozw or something along those lines).

pcpete1234, 2020-10-29:
Thanks for the quick response...not seeing anywhere in my /config

I think it has something to do with how my HA is installed...

This is my config folder...I've searched all sub folders etc

[image]