OpenZWave/open-zwave #2346: Introduce automatic builds for Windows

LordMike, 2020-08-05:
Would it be possible to produce artifacts for Windows in addition to the numerous *nix systems currently?

There are no Windows binaries present here: http://mirror.my-ho.st/Downloads/OpenZWave/ - which makes it harder to consume on Windows machines :)

Mike.

LordMike, 2020-08-05:
I've looked a bit for how to produce the windows binaries using the (old?) vs2010 solution file. I'm not at all well versed in C++.

I've arrived at having an openzwave.lib file, and from [a log for ozw-admin's Windows installer build](http://bamboo.my-ho.st/bamboo/browse/OZW-OZW-161/log), I can see I need to link it somehow to produce the .dll. But the commands are not clear to me.

So I'm pretty much stuck at `msbuild /p:Configuration=Release`

kpine, 2020-08-05:
Perhaps it's `ReleaseDLL` instead of `Release`? Looks like that's how it's built for Windows ozw-admin.

https://github.com/OpenZWave/ozw-admin/blob/da04ebfbadc57484c487ebc846431eb283e83176/scripts/win32build.bat#L3

LordMike, 2020-08-06:
Oh.. Why didn't I look at ozw-admins build script.. 

Thanks!

I now have a DLL :)

But would it still be possible to add it to the regular release schedule? :)