OpenZWave/open-zwave #2347: Controller Backup/Restore?

RicardP, 2020-08-05:
Hi,
Is there any chance to build Backup/Restore function of the Controller into OZW? As I understand it if the Controller goes dead you cannot restore your Nodes into a new Controller!?
A nightmare if I would need to re-include/install/configure my ZWave system even I have only 40 Nodes, but most of them are built in and hard to access... 
Because of that fear I am using Aeotec Gen5 Stick, it supports backup/restore by their special win-program, but it is not very convenient to first stop OZW (Domoticz), run the backup program and finally start Domoticz again... However the real issue for me is that I want to change Controller-brand asap but that brand does not support backup, at least not on Windows :(

Anyway, OZW built in Backup/Restore functionality would be great of course.

Any inputs? :)

ualex73, 2020-09-28:
The backup/restore is USB stick/vendor specific, so building it into openzwave is not possible. Yes, the procedure you describe, works for the Aeotec Gen5 on Windows (contact Aeotec if you like it otherwise)

drzwave, 2020-11-16:
If the USB/Stick is at least a 500 series Z-Wave chip, there are Backup/Restore commands are built into the SerialAPI. See section 7.14 of  INS12350: https://www.silabs.com/documents/public/user-guides/INS12350-Serial-API-Host-Appl.-Prg.-Guide.pdf