OpenZWave/open-zwave #2349: Danalock zwave behavior

lielle1, 2020-08-07:
### Issue Guidelines

Before opening a new issue, please review the following:
* The Github Issues is for OZW Bugs only. Please do not use this for general questions or howto type conversations. If you have a question, please post to our mailing list at https://groups.google.com/forum/#!forum/openzwave
* Please use the search feature to see if your issues have been raised or addressed first.
* Try the OpenZWave Log Analyzer at http://www.openzwave.com/log-analyzer if you are experiencing problems with a device on your network.
* Often we will require Log Files to help diagnose the issue. Please stop any application you are using, remove zwcfg_*.xml/ozwcache_*.xml files, and restart your application. After experiencing your issue, stop your application and attach the OZW_Log.txt file to this issue (please do not filter the log file, we often need the full log file to diagnose issues)
* If you want to add a new device to the OZW Database, please consult https://github.com/OpenZWave/open-zwave/wiki/Adding-Devices
unlike "old" openzwave, current versions won't open my danalock zwave lock if it's already in unlock state. if the state is locked unlock working. 
It is bother me because i have a fixed door handle so i don't need to lock my door all the times.

manznOnly, 2020-08-15:
I have the same problem - in oz 1.4 it works without problems - now it would not open if its in unlock state

lielle1, 2020-08-16:
Yep, It's the only thing keeping from me moving to 1.6 openZwave

MuppetOwl, 2020-08-24:
Can also confirm same problem for Danalock in 1.6. Works with 1.4. 
But for me it about half of the times Open Zwave crash when starting up. Last statuses messages is from the Danalock and then it crash. When it starts up it show as completed. But not possible to lock/unlock as mentioned above. What is strange is that it update correct status when I unlock and lock manually on the lock itself.

slippern, 2020-09-06:
My Danalock is working fine with 1.6
After inclusion, set the gateway in "heal node"-mode and the activate inclusion mode on the lock.
Try several times if its not working, also you can try the same with "refresh config params".

lielle1, 2020-09-07:
> My Danalock is working fine with 1.6
> After inclusion, set the gateway in "heal node"-mode and the activate inclusion mode on the lock.
> Try several times if its not working, also you can try the same with "refresh config params".

Can you unlock the door if it's already unlocked?

slippern, 2020-09-07:
> > My Danalock is working fine with 1.6
> > After inclusion, set the gateway in "heal node"-mode and the activate inclusion mode on the lock.
> > Try several times if its not working, also you can try the same with "refresh config params".
> 
> Can you unlock the door if it's already unlocked?

Yes.