OpenZWave/open-zwave #2354: GoControl WADWAZ-1 Behavior w/ OpenZWave

majorsl, 2020-08-13:
I’m moving over from my old Vera to OpenZWave.  The device is properly identified by OZW (linear/WADWAZ-1.xml).

However, with the Vera it would properly register “Open” continually when it wasn’t in contact with the magnet. With Open Z-Wave it is behaving like a momentary switch, going back to “Closed” after 5 seconds even if the door/window is left open.

I'm more than willing to test a change in the xml file if anyone wants me to try something to solve this.

majorsl, 2020-09-25:
Resolving this.  I found when adding this to Home Assistant, there was an additional disabled entity that was a binary sensor that correctly displayed "open" and "closed" values. This was not apparent, or exposed that I can find, in the ozwadmin app.

closed 2020-09-25 by majorsl

cantes534, 2021-02-25:
Can you advise what sensor you found that corrected displayed the open/closed state correctly? I'm having this issue and have test all the binary sensor available by default for the sensor, only one will show the open / close state but not without the 5 second state reset. Did you need to enable a hidden sensor? 

Any assistance would be appreciated!

majorsl, 2021-02-25:
> Can you advise what sensor you found that corrected displayed the open/closed state correctly? I'm having this issue and have test all the binary sensor available by default for the sensor, only one [...]

All I can remember is it was a disabled one. I’ve since moved on the Zwave-JS and no longer use OZW. I can say that Zwave-JS detects this sensor and correctly reports it to Home Assistant perfectly and it works as expected.