OpenZWave/open-zwave #2375: Re: add support for Iblinds #2353

danielbrunt57, 2020-08-29:
How would one initiate a value refresh for CC 38 from HA?

I have been trying for days to no avail...
My most recent attempt:

`- id: '1598463443954'
  alias: Blinds - Kitchen 1 Poll
  description: '{ "node": 21 } OpenZWave/1/command/refreshValue/'
  trigger:
  - platform: mqtt
    topic: OpenZWave/1/command/setvalue/
  condition: []
  action:
  - delay: 00:00:10
  - data:
      payload: '{ "ValueIDKey": 357122064 }'
      topic: OpenZWave/1/command/refreshvalue/
    service: mqtt.publish
  - delay: 00:00:01
  - data:
      payload: '{ "ValueIDKey": 357138449 }'
      topic: OpenZWave/1/command/refreshvalue/
    service: mqtt.publish
  - delay: 00:00:01
  - data:
      payload: '{ "node": 21 }'
      topic: OpenZWave/1/command/requestnodedynamic/
    service: mqtt.publish
  mode: single`

Fishwaldo, 2020-11-03:
This is a question for HA. Sorry, can't help

closed 2020-11-03 by Fishwaldo

danielbrunt57, 2020-11-03:
FYI, I reverted OZW to version "150" and the command OpenZWave/1/command/refreshvalue/ for ValueID for CC 38 for the node is now working again so IMO it's not an HA issue.  Currently running OpenZWave Version 1.6.1185, qt-openzwave Version: 1.2.0, QT Version: 5.12.8 but in the latest OZW version it's not working...