OpenZWave/open-zwave #2379: Bad node data in z-stick causes z-stick to hang

jefft4, 2020-09-07:
My Z-stick Gen5 has a node that failed to exclude properly and now seems to be a zombie in the z-stick's memory.  I can't get rid of it.  When OZW starts up and discovers all the devices, it's fine up to this node (#34) then the z-stick locks up completely, LED stuck on one colour instead of flashing, and every subsequent node just returns a timeout to OZW.

When I put the z-stick into my PC and run the Zensys z-wave controller app, it's okay - it can't remove node 34 or mark it failed, but the software and the z-stick do not lock up or hit any problems like OZW does and other nodes communicate just fine.  I guess that something in OZW is not handling whatever error this bad node entry causes, and is causing the z-stick to lock up.

All that I get in  OZW_Log.txt on a fresh startup is an error report about node 34, then the z-stick locks up.
`2020-09-07 08:41:01.933 Always, OpenZwave Version 1.6-1275-gf10f80f7 Starting Up
2020-09-07 08:41:01.952 Always, Using Language Localization
2020-09-07 08:41:19.116 Error, Node034, ERROR: Dropping command, expected response not rec
eived after 1 attempt(s). Command: "NoOperation_Set (Node=34): 0x01, 0x09, 0x00, 0x13, 0x2
2, 0x02, 0x00, 0x00, 0x25, 0x27, 0xc7"`

[ozwcache_0xdbf69866.xml.txt](https://github.com/OpenZWave/open-zwave/files/5181711/ozwcache_0xdbf69866.xml.txt)

Note - I may not be able to try further diagnosis unless the response is very quick, as this is a live system so I will need to take the worst-case option and rebuild the z-wave network later today!

jefft4, 2020-09-07:
UPDATE - I've taken a copy of the z-stick EEPROM and restored it to my backup z-stick, which behaves the same way, so should be able to replicate the problem even if I have to rebuild on the primary z-stick.

Fishwaldo, 2020-11-03:
>LED stuck on one colour instead of flashing,

Sounds like a problem with the stick. Factory Reset or raise a issue with Aeotec.

closed 2020-11-03 by Fishwaldo