OpenZWave/open-zwave #2386: OZW Keep communicate with switches then crash

lily148, 2020-09-09:
### Issue Guidelines

I downloaded the open-zwave and compile it myself to use with Domoticz.
I notice, my 2 gangs light switch will keep communicate with the Z-wave stick every second. I am able to reproduce the problem by:

1. Restart the system and run openzwave
2. Let openzwave queried all nodes
3. Turn on the switch then turn it off

Below is the gif:
[image]

Below is the error logs (It keep running until it crash with libc.so.6 and libpthread.so.0:
`Sep 09 12:32:42 smartgateway domoticz[11727]: 2020-09-09 12:32:42.972 Detail, Node025,   Received: 0x01, 0x18, 0x00, 0x13, 0xd5, 0x00, 0x00, 0x0d, 0x02, 0xc6, 0xb8, 0xc7, 0x7f, 0x7f, 0x00, 0x00, 0x03, 0x24, 0x26, 0x00, 0x00, 0x03, 0x01, 0x00, 0x00, 0x94
Sep 09 12:32:42 smartgateway domoticz[11727]: 2020-09-09 12:32:42.973 Detail, Node025,   ZW_SEND_DATA Request with callback ID 0xd5 received (expected 0xd5)
Sep 09 12:32:42 smartgateway domoticz[11727]: 2020-09-09 12:32:42.973 Info, Node025, Request RTT 139 Average Request RTT 137
Sep 09 12:32:42 smartgateway domoticz[11727]: 2020-09-09 12:32:42.974 Detail, Node025, Extended TxStatus: Time: 130, Hops: 2, Rssi:  -58  -72  -57 --- ---, AckChannel: 0, TxChannel: 0, RouteScheme: Last Working Route, Route: 36 38 0 0, RouteSpeed: 100K, RouteTries: 1, FailedLinkFrom: 0, FailedLinkTo: 0
Sep 09 12:32:42 smartgateway domoticz[11727]: 2020-09-09 12:32:42.974 Detail, Node025,   Expected callbackId was received
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.043 Detail, Node025,   Received: 0x01, 0x0f, 0x00, 0x04, 0x00, 0x19, 0x07, 0x60, 0x0d, 0x02, 0x01, 0x25, 0x03, 0x00, 0xc6, 0x00, 0x64
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.043 Detail,
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.043 Info, Node025, Response RTT 210 Average Response RTT 208
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.044 Info, Node025, Received a MultiChannelEncap from node 25, endpoint 2 for Command Class COMMAND_CLASS_SWITCH_BINARY
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.044 Info, Node025, Received SwitchBinary report from node 25: level=Off
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.044 Detail, Node025, Value Updated: old value=true, new value=false, type=bool
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.045 Detail, Node025,         Target Value is Set to true
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.045 Info, Timer: adding event in 250 ms
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.045 Detail, Node025, Value Updated: old value=true, new value=true, type=bool
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.046 Detail, Node025, Changes to this value are not verified
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.046 Detail, Timer: waiting with timeout 37 ms
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.046 Detail, Node025,   Expected reply and command class was received
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.047 Detail, Node025,   Message transaction complete
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.047 Detail,
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.047 Detail, Node025, Removing current message
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.048 Detail, Node025, Notification: ValueRefreshed CC: COMMAND_CLASS_SWITCH_BINARY Instance: 3 Index: 1
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.049 Detail,
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.051 Info, Node025, Sending (Send) message (Callback ID=0xd6, Expected Reply=0x04) - MultiChannel Encapsulated (instance=3): SwitchBinaryCmd_Get (Node=25): 0x01, 0x0d, 0x00, 0x13, 0x19, 0x06, 0x60, 0x0d, 0x01, 0x02, 0x25, 0x02, 0x25, 0xd6, 0x44
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.052 Info, Node025, Encrypted Flag is 0
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.060 Detail, Node025,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.060 Detail, Node025,   ZW_SEND_DATA delivered to Z-Wave stack
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.083 Info, Timer: delayed event
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.083 Info, Node025, Sending Get to Refresh Value after Target Check
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.084 Detail, Node025, Queuing (Send) MultiChannel Encapsulated (instance=3): SwitchBinaryCmd_Get (Node=25): 0x01, 0x0d, 0x00, 0x13, 0x19, 0x06, 0x60, 0x0d, 0x01, 0x02, 0x25, 0x02, 0x25, 0x56, 0xc4
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.084 Detail, Timer: waiting with timeout 211 ms
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.261 Detail, Node025,   Received: 0x01, 0x18, 0x00, 0x13, 0xd6, 0x00, 0x00, 0x14, 0x02, 0xc6, 0xb8, 0xc7, 0x7f, 0x7f, 0x00, 0x00, 0x03, 0x24, 0x26, 0x00, 0x00, 0x03, 0x01, 0x00, 0x00, 0x8e
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.261 Detail, Node025,   ZW_SEND_DATA Request with callback ID 0xd6 received (expected 0xd6)
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.261 Info, Node025, Request RTT 209 Average Request RTT 173
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.262 Detail, Node025, Extended TxStatus: Time: 200, Hops: 2, Rssi:  -58  -72  -57 --- ---, AckChannel: 0, TxChannel: 0, RouteScheme: Last Working Route, Route: 36 38 0 0, RouteSpeed: 100K, RouteTries: 1, FailedLinkFrom: 0, FailedLinkTo: 0
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.262 Detail, Node025,   Expected callbackId was received
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.295 Info, Timer: delayed event
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.295 Info, Node025, Sending Get to Refresh Value after Target Check
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.295 Detail, Node025, Queuing (Send) MultiChannel Encapsulated (instance=3): SwitchBinaryCmd_Get (Node=25): 0x01, 0x0d, 0x00, 0x13, 0x19, 0x06, 0x60, 0x0d, 0x01, 0x02, 0x25, 0x02, 0x25, 0x57, 0xc5
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.295 Detail, Timer: waiting with timeout -1 ms
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.332 Detail, Node025,   Received: 0x01, 0x0f, 0x00, 0x04, 0x00, 0x19, 0x07, 0x60, 0x0d, 0x02, 0x01, 0x25, 0x03, 0x00, 0xc6, 0x00, 0x64
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.332 Detail,
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.332 Info, Node025, Response RTT 280 Average Response RTT 244
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.332 Info, Node025, Received a MultiChannelEncap from node 25, endpoint 2 for Command Class COMMAND_CLASS_SWITCH_BINARY
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.333 Info, Node025, Received SwitchBinary report from node 25: level=Off
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.333 Detail, Node025, Value Updated: old value=true, new value=false, type=bool
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.333 Detail, Node025,         Target Value is Set to true
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.333 Info, Timer: adding event in 250 ms
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.333 Detail, Node025, Value Updated: old value=true, new value=true, type=bool
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.333 Detail, Node025, Changes to this value are not verified
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.333 Detail, Timer: waiting with timeout 249 ms
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.333 Detail, Node025,   Expected reply and command class was received
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.333 Detail, Node025,   Message transaction complete
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.333 Detail,
Sep 09 12:32:43 smartgateway domoticz[11727]: 2020-09-09 12:32:43.333 Detail, Node025, Removing current message
[... cut ...]
Sep 09 12:32:46 smartgateway domoticz[11727]: 2020-09-09 12:32:46.092 Info, Node025, Received a MultiChannelEncap from node 25, endpoint 2 for Command Class COMMAND_CLASS_SWITCH_BINARY
Sep 09 12:32:46 smartgateway domoticz[11727]: 2020-09-09 12:32:46.092 Info, Node025, Received SwitchBinary report from node 25: level=Off
Sep 09 12:33:03 smartgateway domoticz[11727]: 2020-09-09 12:33:03.238  Error: #2  /lib/arm-linux-gnueabihf/libc.so.6 :  + 0x24fe0 [0xb6b93fe0]
Sep 09 12:33:03 smartgateway domoticz[11727]: 2020-09-09 12:33:03.238  Error: #3  /lib/arm-linux-gnueabihf/libpthread.so.0 :  + 0xe346 [0xb6d2c346]`

l3arcf, 2020-09-09:
@lily148 , to me, this appears to be a Binary Switch that may have the same issue as the DZ15S that returns a V1 report when it's V2 capable. I've not been able to confirm this for your device, but if it is true, then my PR (https://github.com/OpenZWave/open-zwave/pull/2370) that fixed (or worked around) my issue could work for you.

Given that you downloaded and compiled yourself, maybe you want to try that PR on your version and give some feedback?  It's maybe a bit of a shot in the dark, but you never know.

My symptoms were the same.  Everything was fine until I turn one of the DZ15S binary switches OFF.  Turning ON had no issue, just OFF was the issue which is how I interpret your comments.

Another qualifier would be whether your switch is V2 compatible and responds with a V1 message.  You can follow my troubleshooting (with insights and direction from @kpine) here: [https://github.com/OpenZWave/qt-openzwave/issues/146](https://github.com/OpenZWave/qt-openzwave/issues/146)

lily148, 2020-09-10:
@l3arcf , your PR (#2370) did help to solve my problem. Yes, my switch is behaving as it is now. Thank you. This PR should be pulled...

closed 2020-09-10 by lily148