OpenZWave/open-zwave #2387: Sunricher SR-ZV9001 4 Channel Dimmer Remote Control+ multichannel issuse,

Micha-123, 2020-09-09:
i have that remote, and i need to controll with it an multichannel Qubino flush relais, 

but if i add Qubino flush 2 relay, it always switch booth lamps on and of, the  node id of my flush relay is 21 so i try to add 21.1 but it stil change simple to 21 and now always both lapms are turned on if i press the button on that remote

that remote supoports 

SUPPORTED COMMAND CLASS
• COMMAND_CLASS_ZWAVEPLUS_INFO_V2
• COMMAND_CLASS_ASSOCIATION_V2
• COMMAND_CLASS_MULTI_CHANNEL_ASSOCIATION_V3
• COMMAND_CLASS_CENTRAL_SCENE_V3
• COMMAND_CLASS_ASSOCIATION_GRP_INFO_V1
• COMMAND_CLASS_TRANSPORT_SERVICE_V2
• COMMAND_CLASS_VERSION_V2
• COMMAND_CLASS_MANUFACTURER_SPECIFIC_V2
• COMMAND_CLASS_DEVICE_RESET_LOCALLY_V1
• COMMAND_CLASS_FIRMWARE_UPDATE_MD_V4
• COMMAND_CLASS_POWERLEVEL_V1
• COMMAND_CLASS_BATTERY_V1
• COMMAND_CLASS_SECURITY_V1
• COMMAND_CLASS_SECURITY_2_V1
• COMMAND_CLASS_WAKE_UP_V2
• COMMAND_CLASS_SUPERVISION_V1

Controlled Command Classes
• COMMAND_CLASS_CENTRAL_SCENE_V3
• COMMAND_CLASS_BASIC_V2
• COMMAND_CLASS_SWITCH_MULTILEVEL_V4
• Multi Channel V3
• Security_V1
• Security_2_V1

so it is supporting multichannel V3 but how i can activate / force it ? to save my node as 21.1 and on other button as 21.2 ?

Micha-123, 2020-09-12:
i finaly get it woirking

im added that lines in srzv9001k12dimz4.xml

</CommandClass>
<CommandClass id="142">
<Compatibility>
<ForceInstances>true</ForceInstances>
</Compatibility>
</CommandClass>

<CommandClass action="remove" id="32"/>
<CommandClass id="96">
<Compatibility>
<MapRootToEndpoint>true</MapRootToEndpoint>
</Compatibility>
</CommandClass>

now it is possible to set groups like 21.1 and 21.2 and it is working

Fishwaldo, 2020-11-03:
Please submit a PR with the fixes. The device database is maintained by the community.

closed 2020-11-03 by Fishwaldo