OpenZWave/open-zwave #2398: Add Leviton VRCS4 missing from manufacturer_specific.xml by referencing VRCZ4 xml

blhoward2, 2020-09-19:
I have two Leviton VRCS4 devices, which are slightly different than the VRCZ4 devices for which an xml already exists. The primary difference is that the VRCS4 is a scene controller, while the VRCZ4 is a zone controller. At present the VRCS4 shows up as an unknown device due to the type identifier being different. As the VRCZ4 xml contains only an association group entry, it should work fine for the VRCS4 as well.

I added a new reference to the VRCS4 pointing to the VRCZ4 xml. There still isn't a ton of functionality in the xml as you have to use the PC Controller software to cause it to send scene_activated events, but at least it will be identified correctly. Could this be merged in going forward?

New entry:

> 
`<Product config="leviton/vrcz4.xml" id="0261" name="VRCS4-M0Z 4-Button Scene Controller" type="0802"/>`

Fishwaldo, 2020-11-03:
Please submit a PR. The device database is maintained by the community.

closed 2020-11-03 by Fishwaldo

blhoward2, 2020-11-03:
I’ll take a look at doing that but you should change the documentation for adding new devices as it presents this as an option:

“ 3 if you had downloaded a copy of our stable/snapshot releases, then just create a new issue on our bug tracker, listing the Manufacturer, Product Type, Product ID, Product Name, and email any configuration file you had created to our mailing list”