OpenZWave/open-zwave #2403: Node values 'Lost' after 'Refresh Node Info' (Fibaro Wall Plug)

kvaster, 2020-09-24:
Recently I have added several Fibaro wall plugs in my system - FGWPE/F Wall Plug Gen5 (FIBARO System).

After initialization it was all OK - I was able to see following items under 'User':

* Switch
* Power
* Electric kWh
* Electric W
* Exporting

But after 'Refresh Node Info' I can see only following items:

* Switch
* Power

I'm using zwave2mqtt with open-zwave.

**open-zwave version:** commit 7eaae21b3248a82f841900a27ec9702d144f53cc

In ozwcache after adding node (before refresh):
```
<CommandClass id="50" name="COMMAND_CLASS_METER">
                                <Compatibility />
                                <State>
                                        <CCVersion>2</CCVersion>
                                        <InNif>true</InNif>
                                        <StaticRequests>2</StaticRequests>
                                </State>
                                <Instance index="1" />
                                <Value type="decimal" genre="user" instance="1" index="0" label="Electric - kWh" units="kWh" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="4.88">
                                        <Help></Help>
                                </Value>
                                <Value type="decimal" genre="user" instance="1" index="2" label="Electric - W" units="W" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="0.0">
                                        <Help></Help>
                                </Value>
                                <Value type="bool" genre="user" instance="1" index="256" label="Exporting" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False">
                                        <Help></Help>
                                </Value>
                                <Value type="button" genre="system" instance="1" index="257" label="Reset" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0">
                                        <Help></Help>
                                </Value>
                        </CommandClass>
```

And in ozwcache after node refresh command:

```
<CommandClass id="50" name="COMMAND_CLASS_METER">
                                <Compatibility />
                                <State>
                                        <InNif>true</InNif>
                                        <StaticRequests>2</StaticRequests>
                                </State>
                                <Instance index="1" />
                        </CommandClass>
```

Logs attached. Full one and filtered. Fibaro plug is node 16.
[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/5277566/OZW_Log.txt)

Filtered:
[OZW_Log.filtered.txt](https://github.com/OpenZWave/open-zwave/files/5277569/OZW_Log.filtered.txt)

davle, 2020-09-30:
I have the same problem with all my fibaro wallplugs after moving to home assistants openzwave beta https://www.home-assistant.io/integrations/ozw/, before I was using the old home assistant zwave implementation that did not have this problem.

RicardP, 2020-09-30:
Seems having the same issue on Domoticz-OZW, Fibaro FGWPE/F only Switch and Power - since I think beginning of July.

[image]

kvaster, 2020-09-30:
I also had no problems with previous versions of open-zwave library, but I can't say when problem started.

davle, 2020-10-12:
Trying to make some sense of this, I took a look at your logs @kvaster.

The first time it is added we can see that we get a call to MeterCmd_SupportedGet, like this:

2020-09-24 18:41:18.598 Info, Node016, Creating ValueIDs for Instance 1 on COMMAND_CLASS_METER
2020-09-24 18:41:18.598 Detail, Node016, Queuing (Query) MeterCmd_SupportedGet (Node=16): 0x01, 0x09, 0x00, 0x13, 0x10, 0x02, 0x32, 0x03, 0x25, 0x6c, 0x8f

The device responds and the values are added:
2020-09-24 18:41:18.739 Detail, Node016,   Expected reply and command class was received
2020-09-24 18:41:18.740 Detail, Node016,   Message transaction complete
2020-09-24 18:41:18.741 Detail,
2020-09-24 18:41:18.742 Detail, Node016, Removing current message
2020-09-24 18:41:18.743 Detail, Node016, Notification: ValueAdded CC: COMMAND_CLASS_METER Instance: 1 Index: 0
2020-09-24 18:41:18.744 Detail, Node016, Notification: ValueAdded CC: COMMAND_CLASS_METER Instance: 1 Index: 2
2020-09-24 18:41:18.746 Detail, Node016, Notification: ValueAdded CC: COMMAND_CLASS_METER Instance: 1 Index: 256
2020-09-24 18:41:18.747 Detail, Node016, Notification: ValueAdded CC: COMMAND_CLASS_METER Instance: 1 Index: 257

On refresh, it looks like this:
2020-09-24 18:42:54.943 Info, Node016, Creating ValueIDs for Instance 1 on COMMAND_CLASS_SENSOR_MULTILEVEL
2020-09-24 18:42:54.944 Detail, Node016, Queuing (Query) SensorMultilevelCmd_Get (Node=16): 0x01, 0x09, 0x00, 0x13, 0x10, 0x02, 0x31, 0x04, 0x25, 0x94, 0x73
2020-09-24 18:42:54.944 Info, Node016, Creating ValueIDs for Instance 1 on COMMAND_CLASS_METER
2020-09-24 18:42:54.943 Info, Node016, Creating ValueIDs for Instance 1 on COMMAND_CLASS_SENSOR_MULTILEVEL
2020-09-24 18:42:54.944 Detail, Node016, Queuing (Query) SensorMultilevelCmd_Get (Node=16): 0x01, 0x09, 0x00, 0x13, 0x10, 0x02, 0x31, 0x04, 0x25, 0x94, 0x73
2020-09-24 18:42:54.944 Info, Node016, Creating ValueIDs for Instance 1 on COMMAND_CLASS_METER
2020-09-24 18:42:54.944 Info, Node016, Creating ValueIDs for Instance 1 on COMMAND_CLASS_CRC_16_ENCAP

COMMAND_CLASS_SENSOR_MULTILEVEL (and the others) gets a query for SensorMultilevelCmd_Get and values are created, but for some reason 
COMMAND_CLASS_METER does not get corresponding query sent, and therefore no values are created.

Maybe someone with some better insight into the codebase can have a look or guide me in the right direction.

/ David

davle, 2020-10-12:
Looking through the code I think I might have found something.
In both cases it requests versions for all the command classes, as part of the queries that it runs.
But when refreshing it doesn't look like the actual request is sent, so the version is not updated (at least not at this stage).

Initial add:
2020-09-24 18:41:17.616 Info, Node016, Requesting Versions for COMMAND_CLASS_METER - Max: 5 - Current 0
2020-09-24 18:41:17.617 Detail, Node016, Setting Encryption Flag on Message For Command Class COMMAND_CLASS_VERSION
2020-09-24 18:41:17.617 Detail, Node016, Queuing (Query) VersionCmd_CommandClassGet (Node=16): 0x01, 0x0a, 0x00, 0x13, 0x10, 0x03, 0x86, 0x13, 0x32, 0x25, 0x60, 0x17

Refresh:
2020-09-24 18:42:54.704 Info, Node016, Requesting Versions for COMMAND_CLASS_SENSOR_MULTILEVEL - Max: 11 - Current 0
2020-09-24 18:42:54.705 Info, Node016, Requesting Versions for COMMAND_CLASS_METER - Max: 5 - Current 0
2020-09-24 18:42:54.705 Info, Node016, Requesting Versions for COMMAND_CLASS_CRC_16_ENCAP - Max: 1 - Current 0

So when RequestState is called on the Meter command class, it doesn't call supportedGet because it only handles versions > 1.

			bool Meter::RequestState(uint32 const _requestFlags, uint8 const _instance, Driver::MsgQueue const _queue)
			{
				bool res = false;
				if (GetVersion() > 1)
				{
					if (_requestFlags & RequestFlag_Static)
					{
						Msg* msg = new Msg("MeterCmd_SupportedGet", GetNodeId(), REQUEST, FUNC_ID_ZW_SEND_DATA, true, true, FUNC_ID_APPLICATION_COMMAND_HANDLER, GetCommandClassId());
						msg->SetInstance(this, _instance);
						msg->Append(GetNodeId());
						msg->Append(2);
						msg->Append(GetCommandClassId());
						msg->Append(MeterCmd_SupportedGet);
						msg->Append(GetDriver()->GetTransmitOptions());
						GetDriver()->SendMsg(msg, _queue);
						res = true;
					}
				}

				if (_requestFlags & RequestFlag_Dynamic)
				{
					res |= RequestValue(_requestFlags, 0, _instance, _queue);
				}

				return res;
			}

kvaster, 2020-10-15:
@davle, thanks! Your investigation helped a lot.

During both node add and node refresh `Driver::InitNode` is called, but during node add we have non-null `protocolInfo`. As a result all command classes will be listed during such initialization:

```
2020-09-24 18:41:17.135 Detail, Node016, Queuing (Command) SecurityCmd_SchemeGet (Node=16): 0x01, 0x0a, 0x00, 0x13, 0x10, 0x03, 0x98, 0x04, 0x00, 0x25, 0x56, 0x1a
2020-09-24 18:41:17.135 Info, Node016,   Optional CommandClasses for node 16:
2020-09-24 18:41:17.136 Info, Node016,     COMMAND_CLASS_ZWAVEPLUS_INFO
2020-09-24 18:41:17.136 Info, Node016,     COMMAND_CLASS_APPLICATION_STATUS
2020-09-24 18:41:17.137 Info, Node016,     COMMAND_CLASS_ASSOCIATION
2020-09-24 18:41:17.137 Info, Node016,   CommandClass 0x59 - NOT REQUIRED
2020-09-24 18:41:17.138 Info, Node016,     COMMAND_CLASS_CONFIGURATION
2020-09-24 18:41:17.138 Info, Node016,     COMMAND_CLASS_CRC_16_ENCAP
2020-09-24 18:41:17.138 Info, Node016,     COMMAND_CLASS_DEVICE_RESET_LOCALLY
2020-09-24 18:41:17.139 Info, Node016,   CommandClass 0x7a - NOT REQUIRED
2020-09-24 18:41:17.139 Info, Node016,     COMMAND_CLASS_MANUFACTURER_SPECIFIC (Existing)
2020-09-24 18:41:17.140 Info, Node016,     COMMAND_CLASS_METER
2020-09-24 18:41:17.140 Info, Node016,     COMMAND_CLASS_MULTI_CHANNEL_ASSOCIATION
2020-09-24 18:41:17.141 Info, Node016,     COMMAND_CLASS_NOTIFICATION
2020-09-24 18:41:17.141 Info, Node016,     COMMAND_CLASS_POWERLEVEL
2020-09-24 18:41:17.141 Info, Node016,     COMMAND_CLASS_SECURITY (Existing)
2020-09-24 18:41:17.142 Info, Node016,     COMMAND_CLASS_SENSOR_MULTILEVEL
2020-09-24 18:41:17.142 Info, Node016,     COMMAND_CLASS_SWITCH_BINARY (Existing)
2020-09-24 18:41:17.143 Info, Node016,     COMMAND_CLASS_VERSION
```

Right after that we begin to query `ProtocolInfo`, `NodeInfo` and `NodePlusInfo`. During `NodeInfo` command class meter is received and some functions are called:

```
Node::SetProtocolInfo
Node::UpdateNodeInfo
Node::SetStaticRequests
```

SetStaticRequests will set flags for GetVersion and GetMultiInstance if and only if apropriate command classes are already registered. And this is the case with node add mode. But in node refresh mode `COMMAND_CLASS_VERSION` is received after that stage - in NodeInfoPlus query.

I've made a primitive patch for this case (it works for me), but I'm not sure if this is a good patch:

```
diff --git a/cpp/src/Node.cpp b/cpp/src/Node.cpp
index 99e1b908..110a9875 100644
--- a/cpp/src/Node.cpp
+++ b/cpp/src/Node.cpp
@@ -1852,6 +1852,7 @@ void Node::SetSecuredClasses(uint8 const* _data, uint8 const _length, uint32 con
                        Log::Write(LogLevel_Info, m_nodeId, "    %s (Unsecured) - %s", it->second->GetCommandClassName().c_str(), it->second->IsInNIF() ? "InNIF" : "NotInNIF");
        }

+       SetStaticRequests();
 }
 //-----------------------------------------------------------------------------
 // <Node::UpdateNodeInfo>
```

kvaster, 2020-10-20:
It seems that device behaviour is really different in secure and non-secure mode. There is no such problem with wallplug when adding it in a non-secure mode.

RicardP, 2020-10-21:
Hi, in my example the plug was included in secure mode and worked properly until, as mentioned above, I estimate begining of July.

PR #2425 (2020-11-03): Update static request flags after receiving all secure command classes
We need to update static request flags only after receiving all secure command classes, cause sometime on node refres command class version is received at the end.

Fixes: #2403
closed 2020-11-03 by Fishwaldo