OpenZWave/open-zwave #2409: Config file downloads fail

kpine, 2020-09-29:
I issued a `checklatestmfsrevision` command to ozwdaemon, and it determined the MFS file was out of date, downloaded it and tried to download several device XML files. Some downloads were successful, others failed. The mfs.xml file was successfully updated.

Logs: [ozwd.log.gz](https://github.com/OpenZWave/open-zwave/files/5302389/ozwd.log.gz)

Fishwaldo, 2020-11-03:
There is still some shaky code on updating the website after commits. I'll look into it again soon.

firstof9, 2020-11-04:
My lock stopped reporting it's status in the past few days, unpon investigation I found that `kwikset/smartcode.xml` contents had some how been written into `kwikset/910.xml`. 

Something weird going on there.

kpine, 2020-11-04:
I confirmed my 910.xml file was the same incorrect file. The entry for that device was changed from smartcode.xml to 910.xml, the 910.xml being a brand new file. I assume it must be related to that.

closed 2022-06-06 by kpine