OpenZWave/open-zwave #2418: Door lock not reporting the status and not responding.

swooshnoypi, 2020-10-08:
Good day! I am using Samsung SHP-DP728 door lock with HMDM100 Module (z-wave plus) attached to it trying to integrate with HA 0.115.6 using open z-wave. My z-wave receiver is Everspring USB z-wave.

Everyspring USB Zwave: https://www.everspring.com/portfolio-item/sa413-z-wave-usb-stick/
Samsung SHP-DP728: https://www.samsungdigitallife.com/shp-dp728/

HA able to detect the device and I could see the entity of the door lock but when trying to click-lock or unlock, no response and no error can be seen from HA. Also, if I tried to physically lock or unlock the door lock it does not report the status from HA.

I've attached the full logs here and the XML file as well. 

I really appreciate and thank you for helping me with this issue.

[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/5346533/OZW_Log.txt)

[config.zip](https://github.com/OpenZWave/open-zwave/files/5346542/config.zip)

swooshnoypi, 2020-10-19:
Hi @l3arcf, many thanks for your advice. Yes, it is a Z-Wave chip which is attached to the Samsung lock. possibly it could be the old USB stick that was previously used. I really hope I could still be able to use it with the USB stick I bought.

Fishwaldo, 2020-11-03:
Looking at the logs, its seems its failing in the encryption/decryption. Maybe it was included with a different Network Key? Please exclude, factory reset and try to include again. 

And yes, 1.4 is depreciated.

closed 2020-11-03 by Fishwaldo

hongkongkiwi, 2021-09-09:
Were you able to resolve this? wondering whether to buy the HMDM100 module.