OpenZWave/open-zwave #2426: Philio PAN06-1 Relays problem after switching from OZW 1.4 to 1.6

quarcko, 2020-10-21:
Hello, i am a Home assistant user of this library, and recently i made a switch to new OZW integration
which uses a "addon" that has OpenZwave 1.6. And my switch PAN06-1 Started to misbehave. (all other devices are fine).

So:
This switch is supposed to have 3 endpoints for control (1,2,3) - 1 and 2 controls each relay separately, and 3rd triggers 
them both. With old 1.4 version of library, something was discovered wrong, as i have 4 switches discovered like this:

<CommandClass id="37" name="COMMAND_CLASS_SWITCH_BINARY" version="1" request_flags="5" innif="true">
<Instance index="1" /> 
<Instance index="2" endpoint="1" />
<Instance index="3" endpoint="2" />
<Instance index="4" endpoint="3" />
.....

BUT - instance 1 & 2 worked correctly, switching 1st and 2nd relay acoordingly, and reporting back their status.

Now with OpenZwave 1.6 this switch is discovered like this: (correctly with 3 swtches):

<CommandClass id="37" name="COMMAND_CLASS_SWITCH_BINARY"> 
<Compatibility /> 
<State> 
<CCVersion>1</CCVersion> 
<InNif>true</InNif> 
<StaticRequests>1</StaticRequests> 
</State> 
<Instance index="1" endpoint="1" /> 
<Instance index="2" endpoint="2" /> 
<Instance index="3" endpoint="3" />
...

So i have correcet count of 3 switches, but something is not right:

1) When i trigger switch 1, it triggers 1st relay and report it as ON (or OFF) - CORRECT
2) When i trigger switch 2, it triggers 2nd relay, BUT reports that BOTH switch 1 and 2 are ON (or OFF)
     But in reality it triggered only the correct relay 2 - Physical action is correct, but report is wrong
3) When i trigger switch 3 - it triggers both relay 1 & 2 - But reports that only switch 3 & 1 are ON (or OFF)
    But in fact all wtiches should be ON (or OFF) - also reporting problem.

Here are some logs:

log when triggering instance 1 to ON:
https://pastebin.com/eijxzd22

log when triggering instance 2 to ON:
https://pastebin.com/Hj0VBeiM

log when triggering instance 3 to ON:
https://pastebin.com/qqQB6S7v

If anything else is needed i will provide, hope it can be solved.
I tried to as Home Assistant support before, but it seems that issue is with OpenZwave itself.

kpine, 2020-10-21:
For proper debugging ideally you would want to a) delete the ozwcache*.xml file, b) restart ozwdaemon, c) provide the log files that show the interview process and provide the resulting ozwcache*.xml file. If you are using the HA OpenZWave Addon, this will be difficult to do. Consider installing the standalone docker container instead, you can do this via the Portainer addon.

Based on your description of OZW 1.4 behavior, for some reason it sounds like the endpoint mapping was not being configured correctly. Since you had 4 instances it sounds like it was including the root device (instance 1) along with the 3 endpoints (instances 2-4). You could try removing the [endpoint mapping](https://github.com/home-assistant/open-zwave/blob/94267fa298c1882f0dc73c0fd08f1f755ba83e83/config/philio/pan06.xml#L39) from the 1.6 device XML and see if that reverts to the OZW 1.4 behavior. You might need to remove the cache file or remove/add the node to see a difference. Again, this is difficult to do with the Addon because you don't have easy access to the config files, but easy with the standalone container.

In all of the logs the last Binary Switch Report is not encapsulated (is it for the root device?), so maybe it is reflecting the value of both relays combined? In that case, the state would affect the endpoint 1 instance since it is mapping root to endpoint 1.

The user manual is really confusing because it says endpoint 1 is the combination of relay 1 and 2, endpoint 2 is relay 1 and endpoint 3 is relay 2. That does not match the behavior you are seeing.

quarcko, 2020-10-22:
So did a test like you asked by removing that line from pan06.xml.
also as a side note: it is not that hard to do with the addon - just make a backup (just this addon)
download & edit files - and restore it back - did the trick perfectly. 

So, long story short: yes, it is now restored as it was behaving with OZW 1.4

Switch 1: Triggers both relays and reports itself as "ON"
Switch 2: Triggers relay 1 and reports that itself and switch 1 are both ON
Switch 3: Triggers relay 2 and reports itself and switch 1 as ON
Switch 4: Triggers both relays and reports that itself and swtich 1 are ON.

So conclusion:
Switch 1 always reports as ON when either relay is turned on and also acts as "switch all" function.
All the rest is good to me.

So i think either this option should be removed from XML and then i can use
Switch 2 & 3 as my switches (and just ignore 1 & 4, i never use them)
and 2 & 3 will  report correct state of relay.
OR
Keep xml as is - but ignore this "not encapsulated" Reports altogether.
Then also things worked good, just that reporting was messed up.

Your thoughts?

kpine, 2020-10-22:
I think the behavior you're describing now sounds correct, and matches the user manual? Switch 2 and 3 match endpoints 1 and 2. Switch 1 is the root device (behavior undocumented), which apparently controls and reflects the status of both relays ("switch all", which is reasonable). Endpoint 3 seems to be the same functionality as the root device. According to the user manual you should be able to disable endpoint 3 (switch 4) completely via a configuration parameter. You would probably have to set it, then remove the node and re-add it. Or it's also just easy enough to ignore it, and in HA you can disable the entities (but you can't hide the devices). Disabling it would perhaps reduce some of the network traffic. I assume you are getting encapsulated messages for endpoint 3 now.

To me, it seems like the config file is just wrong. Maybe _I'm_ wrong, but I thought the root to endpoint mapping should be used when the first endpoint mirrors the root device. That doesn't seem to be the case for this device.

It still confuses me as to why OZW 1.4 was creating all four instances instead of doing the mapping, even with the setting enabled... In the end it doesn't really matter anymore.

quarcko, 2020-10-25:
Yes, so it is "fixed" in my local version. Would be great to fix this xml config here once and for all.
Also: Disabling 4rd endpoint does not work, if you try to edit this config option it always returns back immediately
that it is enabled. Don't know why, but it does not matter anyway. Just a buggy device (as most zwave stuff i tried)
But hey, it works! :)

Fishwaldo, 2020-11-03:
Please submit a PR with the fix for other users.

closed 2020-11-03 by Fishwaldo