OpenZWave/open-zwave #2428: Network is unresponsive, weird routing and lots of lost packages

darkbasic, 2020-10-21:
I'm using a Rapsberry Pi 2 with the RaZberry 2 (EU) zwave controller. I've been using Home Assistant for about 1 year (which uses openzwave 1.4) and I've recently switched to zwave2mqtt, which uses latest openzwave 1.6.
The network always felt slow, unresponsive and with a lot of missed packets, but recently I've decided to try to get it work well while migrating to the latest openzwave. The house is not too big (about 100 square meters all on one floor) with about 30 zwave plus powered devices clustered in 4 different wall boxes, plus another 7 battery powered devices (one in each room). There are also other ~20 powered devices which never got included because auto inclusion didn't work and I was too lazy to include them manually (since the network never worked well I lost most of my motivation).

One of the problems I have is that when I query a device (for example during inclusion or because I removed zwcfg*.xml) such device will miss some of the functionalities: for example a Qubino Flush Dimmer might not be able to report power. Usually restarting openzwave gets the missing pieces back, probably because it has a cache for what it has already discovered plus it requeries the missing ones.

Also, sending commands might take a while and I don't always get full answers (for example turning on a dimmer switch might show an increase in power but might not show the level going up, or vice versa).

Also I get lots of timeouts and weird routes. This is for example what happens when I turn on a switch on a Qubino Flush Dimmer:

```
2020-10-19T19:28:08.658Z z2m:App Zwave api call: setValue [ 3, 37, 1, 0, true ],
2020-10-19 21:28:08.662 Info, Node003, Value::Set - COMMAND_CLASS_SWITCH_BINARY - Switch - 0 - 1 - True,
2020-10-19 21:28:08.663 Info, Node003, SwitchBinary::Set - Setting to On,
2020-10-19 21:28:08.663 Detail, Node003, Queuing (Send) SwitchBinaryCmd_Set (Node=3): 0x01, 0x0a, 0x00, 0x13, 0x03, 0x03, 0x25, 0x01, 0xff, 0x25, 0x83, 0x9b,
2020-10-19 21:28:08.670 Detail, Node003, Queuing (Send) SwitchBinaryCmd_Get (Node=3): 0x01, 0x09, 0x00, 0x13, 0x03, 0x02, 0x25, 0x02, 0x25, 0x84, 0x62,
2020-10-19 21:28:08.670 Detail, ,
2020-10-19 21:28:08.671 Info, Node003, Sending (Send) message (Callback ID=0x83, Expected Reply=0x13) - SwitchBinaryCmd_Set (Node=3): 0x01, 0x0a, 0x00, 0x13, 0x03, 0x03, 0x25, 0x01, 0xff, 0x25, 0x83, 0x9b,
2020-10-19 21:28:08.671 Info, Node003, Encrypted Flag is 0,
2020-10-19 21:28:08.681 Detail, Node003,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8,
2020-10-19 21:28:08.681 Detail, Node003,   ZW_SEND_DATA delivered to Z-Wave stack,
2020-10-19T19:28:08.682Z z2m:Zwave Success zwave api call setValue ,
2020-10-19 21:28:09.517 Detail, Node003,   Received: 0x01, 0x0b, 0x00, 0x04, 0x00, 0x03, 0x03, 0x26, 0x03, 0x63, 0xb5, 0x00, 0x03,
2020-10-19 21:28:09.522 Detail, ,
2020-10-19 21:28:09.522 Info, Node003, Received SwitchMultiLevel report: level=99,
2020-10-19 21:28:09.522 Detail, Node003, Refreshed Value: old value=0, new value=99, type=byte,
2020-10-19 21:28:09.522 Detail, Node003, Changes to this value are not verified,
2020-10-19 21:28:09.523 Detail, Node003, Notification: ValueChanged,
2020-10-19T19:28:09.526Z z2m:Zwave zwave node 3: changed: 38-1-0:Level:0 -> 99,
2020-10-19 21:28:10.938 Detail, Node003,   Received: 0x01, 0x0b, 0x00, 0x04, 0x00, 0x03, 0x03, 0x26, 0x03, 0x63, 0xb0, 0x00, 0x06,
2020-10-19 21:28:10.939 Detail, ,
2020-10-19 21:28:10.939 Info, Node003, Received SwitchMultiLevel report: level=99,
2020-10-19 21:28:10.939 Detail, Node003, Refreshed Value: old value=99, new value=99, type=byte,
2020-10-19 21:28:10.939 Detail, Node003, Changes to this value are not verified,
2020-10-19 21:28:10.940 Detail, Node003, Notification: ValueChanged,
2020-10-19T19:28:10.944Z z2m:Zwave zwave node 3: changed: 38-1-0:Level:99 -> 99,
2020-10-19 21:28:10.976 Detail, Node003,   Received: 0x01, 0x0b, 0x00, 0x04, 0x00, 0x03, 0x03, 0x26, 0x03, 0x63, 0xc1, 0x00, 0x77,
2020-10-19 21:28:10.977 Detail, ,
2020-10-19 21:28:10.977 Info, Node003, Received SwitchMultiLevel report: level=99,
2020-10-19 21:28:10.977 Detail, Node003, Refreshed Value: old value=99, new value=99, type=byte,
2020-10-19 21:28:10.977 Detail, Node003, Changes to this value are not verified,
2020-10-19 21:28:10.977 Detail, Node003, Notification: ValueChanged,
2020-10-19T19:28:10.982Z z2m:Zwave zwave node 3: changed: 38-1-0:Level:99 -> 99,
2020-10-19 21:28:11.043 Detail, Node003,   Received: 0x01, 0x18, 0x00, 0x13, 0x83, 0x00, 0x00, 0xec, 0x03, 0xc3, 0x7f, 0x7f, 0x7f, 0x7f, 0x00, 0x00, 0x05, 0x42, 0x06, 0x08, 0x00, 0x03, 0x06, 0x00, 0x00, 0x17,
2020-10-19 21:28:11.043 Detail, Node003,   ZW_SEND_DATA Request with callback ID 0x83 received (expected 0x83),
2020-10-19 21:28:11.043 Info, Node003, Request RTT 2371 Average Request RTT 2072,
2020-10-19 21:28:11.044 Detail, Node003, Extended TxStatus: Time: 2360, Hops: 3, Rssi:  -61 --- --- --- ---, AckChannel: 0, TxChannel: 0, RouteScheme: Auto Route, Route: 66 6 8 0, RouteSpeed: 100K, RouteTries: 6, FailedLinkFrom: 0, FailedLinkTo: 0,
2020-10-19 21:28:11.044 Detail, Node003,   Expected callbackId was received,
2020-10-19 21:28:11.044 Detail, Node003,   Expected reply was received,
2020-10-19 21:28:11.044 Detail, Node003,   Message transaction complete,
2020-10-19 21:28:11.044 Detail, ,
2020-10-19 21:28:11.044 Detail, Node003, Removing current message,
2020-10-19 21:28:11.045 Detail, ,
2020-10-19 21:28:11.045 Info, Node003, Sending (Send) message (Callback ID=0x84, Expected Reply=0x04) - SwitchBinaryCmd_Get (Node=3): 0x01, 0x09, 0x00, 0x13, 0x03, 0x02, 0x25, 0x02, 0x25, 0x84, 0x62,
2020-10-19 21:28:11.045 Info, Node003, Encrypted Flag is 0,
2020-10-19 21:28:11.052 Detail, Node003,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8,
2020-10-19 21:28:11.053 Detail, Node003,   ZW_SEND_DATA delivered to Z-Wave stack,
2020-10-19 21:28:11.487 Detail, Node003,   Received: 0x01, 0x0b, 0x00, 0x04, 0x00, 0x03, 0x03, 0x26, 0x03, 0x63, 0xc4, 0x00, 0x72,
2020-10-19 21:28:11.488 Detail, ,
2020-10-19 21:28:11.488 Info, Node003, Response RTT 442 Average Response RTT 824,
2020-10-19 21:28:11.488 Info, Node003, Received SwitchMultiLevel report: level=99,
2020-10-19 21:28:11.488 Detail, Node003, Refreshed Value: old value=99, new value=99, type=byte,
2020-10-19 21:28:11.488 Detail, Node003, Changes to this value are not verified,
2020-10-19 21:28:11.489 Detail, Node003, Notification: ValueChanged,
2020-10-19T19:28:11.493Z z2m:Zwave zwave node 3: changed: 38-1-0:Level:99 -> 99,
2020-10-19 21:28:11.551 Detail, Node003,   Received: 0x01, 0x0b, 0x00, 0x04, 0x00, 0x03, 0x03, 0x26, 0x03, 0x63, 0xc3, 0x00, 0x75,
2020-10-19 21:28:11.552 Detail, ,
2020-10-19 21:28:11.552 Info, Node003, Response RTT 506 Average Response RTT 665,
2020-10-19 21:28:11.552 Info, Node003, Received SwitchMultiLevel report: level=99,
2020-10-19 21:28:11.552 Detail, Node003, Refreshed Value: old value=99, new value=99, type=byte,
2020-10-19 21:28:11.552 Detail, Node003, Changes to this value are not verified,
[... 537 lines cut ...]
2020-10-19 21:54:35.429 Detail, Node046, Notification: RequestNodeNeighborUpdate ControllerCommand - Starting,
2020-10-19 21:54:35.430 Info, Node046, Sending (Command) message (Callback ID=0xdb, Expected Reply=0x48) - ControllerCommand_RequestNodeNeighborUpdate (Node=46): 0x01, 0x05, 0x00, 0x48, 0x2e, 0xdb, 0x47,
  help: 'RequestNodeNeighborUpdate Starting',
2020-10-19 21:54:35.529 Detail, Node046, Notification: RequestNodeNeighborUpdate ControllerCommand - InProgress,
  help: 'RequestNodeNeighborUpdate InProgress',
2020-10-19 21:54:46.145 Detail, Node046, Notification: RequestNodeNeighborUpdate ControllerCommand - Completed,
  help: 'RequestNodeNeighborUpdate Completed',
```

After the heal attempt I removed zwcfg_*.xml/ozwcache_*.xml and restarted the zwave2mqtt container in orde to give the log analyzer a try (I really don't know why it says that zwcfg*.xml was present when the log was collected): http://www.openzwave.com/log-analyzer/public/dbc2e020-1329-45d2-9a84-e4d82b94c92c/

This is the OZW_Log.txt: 
[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/5416984/OZW_Log.txt)

I also think that openzwave crashes because I didn't touch it since yesterday and today the OZW_Log was starting with the following (docker container is up since 25 hours): `2020-10-21 14:09:28.966 Always, OpenZwave Version 1.6-974-ge89d2a16-dirty Starting Up`
For sure yesterday it crashed once while trying to remove a failed node.

[... cut ...]
Also some nodes look dead, but most of the times it doesn't let me remove them. When I finally got it to remove those failed nodes they showed up again on restart. What can I do to truly remove them? I'm not even sure which ones they are: 87 should be a copy of node 89 (it didn't work well so I tried to exclude and reinclude it), 52 should be a copy of one of my 7 battery powered devices (not sure what happened) and 16 could be a broken Qubino Flush dimmer (I miss one light in the list).

Poll interval is 60000 ms in zwave2mqtt.

Please help me to get it working, I'm really tired of this network and I'm giving up for sure unless someone manages to help me debug this nightmare :pray:

robertsLando, 2020-10-21:
Hi Nicolò. Based on my experience this kind of issues can only be resolved in one way: exclude all devices, hard reset the controller and start including back one by one them to the controller. The only person that could help you debugging this is @Fishwaldo but I don't hear since some months ago. Also when you switched from ozw 1.4 to 1.6 did you do this? I mean did you reset your controller and start with a new configuration? 

Poll interval is ok as it should be not less then 1s for device in the network. 

Another solution could be to split the network in 2 smaller ones using 2 controllers

darkbasic, 2020-10-21:
No I didn't... if an hard reset is really the only solution I'll try it, but that would mean that zwave is truly beyond any hope :(

kpine, 2020-10-21:
Does the same problem happen with other nodes besides node 3? Have you tried removing node 3 to see if is causing problems for the whole network? In the first log it looks like node 3 is spamming the network with unsolicited SwitchMultiLevel reports (unless there is something missing from that logs that happened earlier). OZW only requested a single Get for the Binary Switch value and never got a response for that, meanwhile there were something like 16 duplicate reports for the MultiLevel value from the node within 4 seconds.

You probably should try to remove the dead nodes, I think (?) those can cause problems with network routing. With ozwdaemon you would issue a `hasnodefailed` command then followed by a `removefailednode` command, not sure how to do that with zwave2mqtt. As a last resort, you can use the Silicon Labs PC Controller and remove the dead nodes there.

darkbasic, 2020-10-21:
It happens with other nodes as well (node 7 is also a few centimeters away from the controller):
```
2020-10-21T17:20:01.241Z z2m:App Zwave api call: setValue [ 7, 37, 1, 0, true ],
2020-10-21 19:20:01.268 Info, Node007, Value::Set - COMMAND_CLASS_SWITCH_BINARY - Switch - 0 - 1 - True,
2020-10-21 19:20:01.268 Info, Node007, SwitchBinary::Set - Setting to On,
2020-10-21 19:20:01.269 Detail, Node007, Queuing (Send) SwitchBinaryCmd_Set (Node=7): 0x01, 0x0a, 0x00, 0x13, 0x07, 0x03, 0x25, 0x01, 0xff, 0x25, 0x73, 0x6f,
2020-10-21 19:20:01.269 Detail, Node007, Queuing (Send) SwitchBinaryCmd_Get (Node=7): 0x01, 0x09, 0x00, 0x13, 0x07, 0x02, 0x25, 0x02, 0x25, 0x74, 0x96,
2020-10-21 19:20:01.269 Detail, ,
2020-10-21 19:20:01.270 Info, Node007, Sending (Send) message (Callback ID=0x73, Expected Reply=0x13) - SwitchBinaryCmd_Set (Node=7): 0x01, 0x0a, 0x00, 0x13, 0x07, 0x03, 0x25, 0x01, 0xff, 0x25, 0x73, 0x6f,
2020-10-21 19:20:01.270 Info, Node007, Encrypted Flag is 0,
2020-10-21T17:20:01.270Z z2m:Zwave Success zwave api call setValue ,
2020-10-21 19:20:01.276 Detail, Node007,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8,
2020-10-21 19:20:01.277 Detail, Node007,   ZW_SEND_DATA delivered to Z-Wave stack,
2020-10-21 19:20:01.413 Detail, Node007,   Received: 0x01, 0x18, 0x00, 0x13, 0x73, 0x00, 0x00, 0x0d, 0x02, 0xda, 0x7f, 0x7f, 0x7f, 0x7f, 0x00, 0x00, 0x03, 0x4f, 0x11, 0x00, 0x00, 0x03, 0x01, 0x00, 0x00, 0x0d,
2020-10-21 19:20:01.413 Detail, Node007,   ZW_SEND_DATA Request with callback ID 0x73 received (expected 0x73),
2020-10-21 19:20:01.413 Info, Node007, Request RTT 142 Average Request RTT 173,
2020-10-21 19:20:01.413 Detail, Node007, Extended TxStatus: Time: 130, Hops: 2, Rssi:  -38 --- --- --- ---, AckChannel: 0, TxChannel: 0, RouteScheme: Last Working Route, Route: 79 17 0 0, RouteSpeed: 100K, RouteTries: 1, FailedLinkFrom: 0, FailedLinkTo: 0,
2020-10-21 19:20:01.413 Detail, Node007,   Expected callbackId was received,
2020-10-21 19:20:01.414 Detail, Node007,   Expected reply was received,
2020-10-21 19:20:01.414 Detail, Node007,   Message transaction complete,
2020-10-21 19:20:01.414 Detail, ,
2020-10-21 19:20:01.414 Detail, Node007, Removing current message,
2020-10-21 19:20:01.414 Detail, ,
2020-10-21 19:20:01.414 Info, Node007, Sending (Send) message (Callback ID=0x74, Expected Reply=0x04) - SwitchBinaryCmd_Get (Node=7): 0x01, 0x09, 0x00, 0x13, 0x07, 0x02, 0x25, 0x02, 0x25, 0x74, 0x96,
2020-10-21 19:20:01.415 Info, Node007, Encrypted Flag is 0,
2020-10-21 19:20:01.421 Detail, Node007,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8,
2020-10-21 19:20:01.421 Detail, Node007,   ZW_SEND_DATA delivered to Z-Wave stack,
2020-10-21 19:20:01.703 Detail, Node007,   Received: 0x01, 0x0b, 0x00, 0x04, 0x00, 0x07, 0x03, 0x26, 0x03, 0x63, 0xda, 0x00, 0x68,
2020-10-21 19:20:01.703 Detail, ,
2020-10-21 19:20:01.704 Info, Node007, Response RTT 288 Average Response RTT 331,
2020-10-21 19:20:01.704 Info, Node007, Received SwitchMultiLevel report: level=99,
2020-10-21 19:20:01.704 Detail, Node007, Refreshed Value: old value=0, new value=99, type=byte,
2020-10-21 19:20:01.704 Detail, Node007, Changes to this value are not verified,
2020-10-21 19:20:01.704 Detail, Node007, Notification: ValueChanged,
2020-10-21T17:20:01.709Z z2m:Zwave zwave node 7: changed: 38-1-0:Level:0 -> 99,
2020-10-21 19:20:01.780 Detail, Node007,   Received: 0x01, 0x0b, 0x00, 0x04, 0x00, 0x07, 0x03, 0x26, 0x03, 0x63, 0xda, 0x00, 0x68,
2020-10-21 19:20:01.782 Detail, ,
2020-10-21 19:20:01.785 Info, Node007, Response RTT 370 Average Response RTT 350,
2020-10-21 19:20:01.787 Info, Node007, Received SwitchMultiLevel report: level=99,
2020-10-21 19:20:01.789 Detail, Node007, Refreshed Value: old value=99, new value=99, type=byte,
2020-10-21 19:20:01.790 Detail, Node007, Changes to this value are not verified,
2020-10-21 19:20:01.790 Detail, Node007, Notification: ValueChanged,
2020-10-21T17:20:01.795Z z2m:Zwave zwave node 7: changed: 38-1-0:Level:99 -> 99,
2020-10-21 19:20:01.854 Detail, Node007,   Received: 0x01, 0x0b, 0x00, 0x04, 0x00, 0x07, 0x03, 0x26, 0x03, 0x63, 0xda, 0x00, 0x68,
2020-10-21 19:20:01.858 Detail, ,
2020-10-21 19:20:01.861 Info, Node007, Response RTT 445 Average Response RTT 397,
2020-10-21 19:20:01.862 Info, Node007, Received SwitchMultiLevel report: level=99,
2020-10-21 19:20:01.862 Detail, Node007, Refreshed Value: old value=99, new value=99, type=byte,
2020-10-21 19:20:01.863 Detail, Node007, Changes to this value are not verified,
2020-10-21 19:20:01.865 Detail, Node007, Notification: ValueChanged,
2020-10-21T17:20:01.866Z z2m:Zwave zwave node 7: changed: 38-1-0:Level:99 -> 99,
2020-10-21 19:20:02.166 Detail, Node007,   Received: 0x01, 0x18, 0x00, 0x13, 0x74, 0x00, 0x00, 0x4b, 0x02, 0xbb, 0x7f, 0x7f, 0x7f, 0x7f, 0x00, 0x01, 0x04, 0x04, 0x03, 0x00, 0x00, 0x01, 0x02, 0x00, 0x00, 0x73,
2020-10-21 19:20:02.169 Detail, Node007,   ZW_SEND_DATA Request with callback ID 0x74 received (expected 0x74),
2020-10-21 19:20:02.172 Info, Node007, Request RTT 758 Average Request RTT 465,
2020-10-21 19:20:02.175 Detail, Node007, Extended TxStatus: Time: 750, Hops: 2, Rssi:  -69 --- --- --- ---, AckChannel: 0, TxChannel: 1, RouteScheme: Next to Last Working Route, Route: 4 3 0 0, RouteSpeed: 9600, RouteTries: 2, FailedLinkFrom: 0, FailedLinkTo: 0,
2020-10-21 19:20:02.176 Detail, Node007,   Expected callbackId was received,
[... cut ...]
2020-10-21 19:20:11.416 Detail, Node007, Removing current message,
2020-10-21 19:20:11.417 Detail, Node007, Notification: Notification - TimeOut,
2020-10-21T17:20:11.420Z z2m:Zwave Notification from node 7: Notification - TimeOut (1),
```

I tried to remove the dead nodes, but most of the times `hasnodefailed` returns that those nodes are ok. Sometimes I managed to return "failed" and I didn't had to use `removefailednode` because it removed them automatically. Unfortunately as soon as I restarted openzwave the nodes were back again :cry:

robertsLando, 2020-10-22:
> Another weird thing I noticed is that most of my nodes don't show up as "secure" (yes, I'm using the same network key I was using in Home Assistant before):

This is a know bug, AFAIK (after lot of discussions about this with ozw author) the only way to identify security in devices is to check for a specific command class, unfortunally seems not all devices support this class :confused:

Fishwaldo, 2020-11-03:
Looking at the logs, you have some "pre-zwave+" devices in your network, and you have a *LOT* of devices as well. 

pre-zwave+ operates at 40K speed (versus 100K) and at least in some of the log snippets above, Z-Wave is routing through those nodes... so you probably have bandwidth issues. 

As @kpine mentioned, Dead Nodes can play havok. You need to remove them. Also, polling is generally bad (especially if you have low bandwidth) so you should consult devices that allow "instance updates" or lifeline associations. Also, check associations to make sure the control is in the right association groups (some groups on devices will send duplicate information to devices in different associations groups - So if the controller is in multiple groups, and each group sends the same updates - then OZW gets the same update multiple time.

Finally, pre-zwave+ devices are oooooold... They will have a effect if you have a large network. You may wish to look at upgrading them. 

Security should only be enabled on devices that are sensative (door locks etc) as it introduces a fair bit of overhead traffic wise and if you are having bandwidth issues, thats going to make the issue worse.

closed 2020-11-03 by Fishwaldo

darkbasic, 2020-11-03:
Hi @Fishwaldo 
*All* of my powered devices are zwave+ (there is a zwave+ label on them, plus I've bought them on 2019). The only ones which **could** (not 100% sure) be pre-zwave+ are the 7 battery powered Popp Radiator Thermostats (they're basically Danfoss with a custom firmware which reports the temperature) and I'm not aware of better alternatives.
Also I've finally managed to remove the dead nodes, so those shouldn't be an issue anymore.
Do you suggest to open a new issue? Thanks

darkbasic, 2020-11-03:
@robertsLando confirmed that since the table in "Gateway values" was empty I didn't poll any value: poll interval was set but basically not used.