OpenZWave/open-zwave #2429: Log file no longer compatible with log analyzer

GeekGoneOld, 2020-10-21:
Many places I see reference to the log analyzer including in the README.md and in the instructions for entering this issue.

I have tried the analyzer and it fails to analyze.  I believe I have successfully determined that the log format has changed (e.g. putting time and severity in "[]", a comma changed to a colon and, in the case of docker image, some inappropriate entries (such as mqtt).

I would be happy to look at the parser for the analyzer but I can't find the source.  If someone can point me to the source I will try to make it bilingual (definitely the best way).  

Otherwise, we need a conversion program or script which I am happy to work on.  My questions are: 
1. how to distribute (e.g. on the analyzer page may be best)?
2. have extra log entries been created that the analyzer would choke on?

Just give me a little nudge in the right direction and I will help out because I can see this being a useful tool again.

darkbasic, 2020-10-22:
I guess you're using zwave2mqtt, if so be sure to get the log from `/usr/share/hassio/addons/data/*_zwave2mqtt/OZW_Log.txt`
I tried yesterday and it works with openzwave 1.6.974 (zwave2mqtt 4.0.4)

GeekGoneOld, 2020-10-22:
Almost.  I'm actually using the OpenZWave (Beta) integration in HA with an image install on RPi (HassOS, docker supervisor and docker addons).  I have looked and looked for the OZW_Log.txt in the docker and have not found it.  The only log I have found is the docker log.  That is the one that is incompatible.

darkbasic, 2020-10-22:
You can't use the docker log, you have to attach a shell to your container and search for the log in the filesystem (`find / | grep OZW_Log`). Most probably it's binded into a volume so you will have to simply search into the host filesystem (`find /usr/share/hassio | grep OZW_Log`) without the need to attach a  shell to the container.

kpine, 2020-10-22:
`ozwdaemon` (via qt-openzwave) does not generate the OZW_Log.txt file. It overrides the OZW library's default file logger and uses the QT Logging framework instead, which sends logs to stdout. QT has it's own logging format, and the logs also include other components within ozqdaemon (e.g. mqtt), so the output is incompatible with the analyzer, which assumes the default file-based format.

The HA addon doesn't configure any kind of special logging, so I believe stdout is captured in the docker logs, which I guess is visible in the Supervisor logs. The recent versions of the ozwdaemon docker container uses a supervisor process (s6-overlay) that handles saving the logs to files and does log rotations.

At this time you would need to run some kind of parser to transform the log file to the expected format. This same discussion is being had in the mailing list: https://groups.google.com/u/1/g/openzwave/c/bNL4jd5lrGA

GeekGoneOld, 2020-10-22:
That's almost exactly what I had assumed.  The group discussion was started
by me and I have hit a wall wrt finding the source for the log analyzer or
finding what format or what I should/should not  include in the transformed
log file.  Some is obvious, some is not.

Fishwaldo, 2020-11-03:
The log analyzer on the website needs to be updated, but it wont work with anything output from qt-ozwdaemon etc.

closed 2020-11-03 by Fishwaldo

GeekGoneOld, 2020-11-05:
>>>Justin wrote:
>>>The log analyzer on the website needs to be updated, but it wont work
with anything output from qt-ozwdaemon etc.

Glad to help with that if I can.  Is the analyzer open-source?  If so, let
me know where and I will try to make it "bilingual".

Keith