OpenZWave/open-zwave #2434: add Thermofloor heatit Z-temp2

jithurbide-CPNV, 2020-10-27:
Hello,

Cloud you please add this product to openzwave : 

<Product config="thermofloor/heatit204.xml" id="0204" name="Heatit Thermostat Z-Temp2" type="0004"/>.

I have send the config file, images and modified manufacturer_specific.xml to the mailing list. I run xmltest on the file and test it on my computer. Seems to works correctly.

Best regards,

Julien Ithurbide

Fishwaldo, 2020-11-03:
Please open a PR if its still missing. The database is community maintained.

closed 2020-11-03 by Fishwaldo

rakaandro, 2020-12-16:
Hi Julien,

I do need your xml files for my Temp2. Please do a PullRequest with your files. Thanks

jithurbide-CPNV, 2020-12-17:
Hello,

I never do a pull request. Just do it now. I hope it's working.

Best regards,

Julien

On Wed, 2020-12-16 at 11:38 -0800, rakaandro wrote:

Hi Julien,

I do need your xml files for my Temp2. Please do a PullRequest with your files. Thanks

—

————————————————————————————————————————————————————————————
Julien Ithurbide, enseignant de la filière d’informatique
CPNV Centre professionnel du Nord vaudois
Etat de Vaud - Département de la formation,
de la jeunesse et de la culture
Avenue de la Gare 14, CH-1450 Sainte-Croix
Mobile : 079 255 67 08
mailto:Julien.ithurbide@cpnv.ch<mailto:Julien.ithurbide@cpnv.ch>

jithurbide-CPNV, 2020-12-17:
@Fishwaldo Could you please tell me if my pull request is correct ?