OpenZWave/open-zwave #2436: AEON Labs ZW090 Z-Stick Gen5 ZWave Plus Flag

Veldkornet, 2020-10-27:
I have a [AEON Labs ZW090-C Z-Stick Gen5 EU](https://products.z-wavealliance.org/products/1355), which support ZWave Plus; however according to OZWAdmin, ZWave Plus is not enabled.

I am currently using the [OpenZWave of HomeAssistant](https://github.com/home-assistant/hassio-addons/tree/master/zwave)

[image]

[image]

kpine, 2020-10-27:
This is the current behavior for all Z-Wave Plus controllers. Controllers do not expose this flag the way other nodes do so it is no reported correctly.

This is a duplicate of #2129 and has already been fixed in the Dev branch. The fix will not be available until the next OZW 1.7 release.

Veldkornet, 2020-10-27:
Ah okay, thanks!

closed 2020-10-27 by Veldkornet