OpenZWave/open-zwave #2448: OZW not receiving updates from nodes

chrisdrackett, 2020-11-03:
I'm having an issue where I'm able to send commands to all my nodes via the OZWAdmin without issue, but if I change a node directly (flip a switch for example) the update does not make it back into the system.

Here is what I see in the logs when flipping a switch (node 28) physically:

```
[20201103 11:32:32.066 PST] [default] [debug]: QVariant(int, 9) 
[20201103 11:32:32.066 PST] [default] [debug]: QVariant(int, 10) 
[20201103 11:32:32.067 PST] [default] [debug]: QVariant(int, 11) 
[20201103 11:32:32.067 PST] [default] [debug]: QVariant(int, 12) 
[20201103 11:32:32.068 PST] [default] [debug]: QVariant(int, 13) 
[20201103 11:32:32.069 PST] [default] [debug]: QVariant(int, 15) 
[20201103 11:32:32.070 PST] [default] [debug]: QVariant(int, 16) 
[20201103 11:32:32.071 PST] [default] [debug]: QVariant(int, 17) 
[20201103 11:32:32.072 PST] [default] [debug]: QVariant(int, 19) 
[20201103 11:32:32.073 PST] [default] [debug]: QVariant(int, 20) 
[20201103 11:32:32.074 PST] [default] [debug]: QVariant(int, 22) 
[20201103 11:32:32.074 PST] [default] [debug]: QVariant(int, 23) 
[20201103 11:32:32.075 PST] [default] [debug]: QVariant(int, 24) 
[20201103 11:32:32.076 PST] [default] [debug]: QVariant(int, 25) 
[20201103 11:32:32.076 PST] [default] [debug]: QVariant(int, 26) 
[20201103 11:32:32.077 PST] [default] [debug]: QVariant(int, 27) 
[20201103 11:32:32.078 PST] [default] [debug]: QVariant(int, 29) 
[20201103 11:32:32.079 PST] [default] [debug]: QVariant(int, 32) 
[20201103 11:32:32.079 PST] [default] [debug]: QVariant(int, 33) 
[20201103 11:32:32.080 PST] [default] [debug]: QVariant(int, 34) 
[20201103 11:32:32.081 PST] [default] [debug]: QVariant(int, 35) 
[20201103 11:32:32.081 PST] [default] [debug]: QVariant(int, 36) 
[20201103 11:32:32.082 PST] [default] [debug]: QVariant(int, 37) 
[20201103 11:32:32.082 PST] [default] [debug]: QVariant(int, 38) 
[20201103 11:32:32.083 PST] [default] [debug]: QVariant(int, 39) 
[20201103 11:32:32.083 PST] [default] [debug]: QVariant(int, 40) 
[20201103 11:32:32.084 PST] [default] [debug]: QVariant(int, 41) 
[20201103 11:32:32.084 PST] [default] [debug]: QVariant(int, 42) 
[20201103 11:32:32.085 PST] [default] [debug]: QVariant(int, 43) 
[20201103 11:32:32.085 PST] [default] [debug]: QVariant(int, 30) 
[20201103 11:32:32.085 PST] [default] [debug]: QVariant(int, 31) 
[20201103 11:32:32.086 PST] [default] [debug]: QVariant(int, 28) 
```

as far as I can tell there isn't anything to see.

and here is an example of my logs when I change a value via OZWAdmin (still node 28):

```
[20201103 11:31:28.680 PST] [ozw.library] [info]: Info - Node: 28 Response RTT 137 Average Response RTT 137 
[20201103 11:31:28.680 PST] [ozw.logging] [debug]: popping Log Mesages 
[20201103 11:31:28.685 PST] [ozw.library] [info]: Info - Node: 28 Received SwitchMultiLevel report: level=18 
[20201103 11:31:28.685 PST] [ozw.logging] [debug]: popping Log Mesages 
[20201103 11:31:28.689 PST] [ozw.library] [debug]: Detail - Node: 28 Value Updated: old value=99, new value=18, type=byte 
[20201103 11:31:28.689 PST] [ozw.logging] [debug]: popping Log Mesages 
[20201103 11:31:28.692 PST] [ozw.library] [debug]: Detail - Node: 28 Changes to this value are verified 
[20201103 11:31:28.692 PST] [ozw.logging] [debug]: popping Log Mesages 
[20201103 11:31:28.695 PST] [ozw.library] [info]: Info - Node: 28 Changed value (changed again)--rechecking 
[20201103 11:31:28.696 PST] [ozw.logging] [debug]: popping Log Mesages 
[20201103 11:31:28.698 PST] [ozw.library] [info]: Info - Node: 0 mgr,     Refreshing node 28: COMMAND_CLASS_SWITCH_MULTILEVEL index = 0 instance = 1 (to confirm a reported change) 
[20201103 11:31:28.699 PST] [ozw.logging] [debug]: popping Log Mesages 
[20201103 11:31:28.701 PST] [ozw.library] [debug]: Detail - Node: 28 Queuing (Send) SwitchMultilevelCmd_Get (Node=28): 0x01, 0x09, 0x00, 0x13, 0x1c, 0x02, 0x26, 0x02, 0x25, 0x95, 0x6f 
[20201103 11:31:28.702 PST] [ozw.logging] [debug]: popping Log Mesages 
[20201103 11:31:28.704 PST] [ozw.library] [debug]: Detail - Node: 28   Expected reply and command class was received 
[20201103 11:31:28.705 PST] [ozw.logging] [debug]: popping Log Mesages 
[20201103 11:31:28.707 PST] [ozw.library] [debug]: Detail - Node: 28   Message transaction complete 
[20201103 11:31:28.707 PST] [ozw.logging] [debug]: popping Log Mesages 
[20201103 11:31:28.710 PST] [ozw.library] [debug]: Detail - Node: 28 Removing current message 
[20201103 11:31:28.710 PST] [ozw.logging] [debug]: popping Log Mesages 
[20201103 11:31:28.713 PST] [ozw.library] [info]: Info - Node: 28 Sending (Send) message (Callback ID=0x95, Expected Reply=0x04) - SwitchMultilevelCmd_Get (Node=28): 0x01, 0x09, 0x00, 0x13, 0x1c, 0x02, 0x26, 0x02, 0x25, 0x95, 0x6f 
[20201103 11:31:28.713 PST] [ozw.logging] [debug]: popping Log Mesages 
[20201103 11:31:28.716 PST] [ozw.library] [info]: Info - Node: 28 Encrypted Flag is 0 
[20201103 11:31:28.716 PST] [ozw.logging] [debug]: popping Log Mesages 
[20201103 11:31:28.726 PST] [ozw.library] [debug]: Detail - Node: 28   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8 
[20201103 11:31:28.726 PST] [ozw.logging] [debug]: popping Log Mesages 
[... 127 lines cut ...]
[20201103 11:34:24.827 PST] [default] [debug]: QVariant(QString, "Office Light") 
[20201103 11:34:24.827 PST] [default] [debug]: QVariant(QString, "") 
[20201103 11:34:24.827 PST] [default] [debug]: QVariant(QString, "GE (Jasco Products)") 
[20201103 11:34:24.828 PST] [default] [debug]: QVariant(QString, "12724 3-Way Dimmer Switch") 
[20201103 11:34:24.828 PST] [default] [debug]: QVariant(QString, "Routing Slave") 
[20201103 11:34:24.829 PST] [default] [debug]: QVariant(int, 4) 
[20201103 11:34:24.829 PST] [default] [debug]: QVariant(QString, "Multilevel Switch") 
[20201103 11:34:24.830 PST] [default] [debug]: QVariant(int, 17) 
[20201103 11:34:24.830 PST] [default] [debug]: QVariant(QString, "Multilevel Power Switch") 
[20201103 11:34:24.830 PST] [default] [debug]: QVariant(int, 1) 
[20201103 11:34:24.830 PST] [default] [debug]: QVariant(QString, "Unknown Type (0x0000)") 
[20201103 11:34:24.831 PST] [default] [debug]: QVariant(int, 0) 
[20201103 11:34:24.831 PST] [default] [debug]: QVariant(int, 0) 
[20201103 11:34:24.831 PST] [default] [debug]: QVariant(QString, "Central Controller") 
[20201103 11:34:24.832 PST] [default] [debug]: QVariant(int, 0) 
[20201103 11:34:24.832 PST] [default] [debug]: QVariant(QString, "Z-Wave+ node") 
[20201103 11:34:24.832 PST] [default] [debug]: QVariant(QString, "Complete") 
[20201103 11:34:24.832 PST] [default] [debug]: QVariant(QString, "0x0063") 
[20201103 11:34:24.833 PST] [default] [debug]: QVariant(QString, "0x4944") 
[20201103 11:34:24.833 PST] [default] [debug]: QVariant(QString, "0x3031") 
[20201103 11:34:24.833 PST] [default] [debug]: QVariant(uint, 40000) 
[20201103 11:34:24.834 PST] [default] [debug]: QVariant(int, 4) 
[20201103 11:34:24.834 PST] [default] [debug]: QVariant(int, 0) 
[20201103 11:34:24.834 PST] [default] [debug]: QVariant(QBitArray, QBitArray(1011 0011 0)) 
```

MattiasC, 2020-11-17:
I am having the exact same problem with a Qubino ZMNHDD1 Flush Dimmer. I can control it via HomeAssistant and can see the messages published to the MQTT broker, but when physically flipping the switch, no messages are published to the MQTT broker and nothing happens in the log (tail -f ozw/logs/ozwdaemon/current). I think this issue is connected to https://github.com/home-assistant/core/issues/40187

I am not very experienced in debugging this, but if there is any more information needed, let me know and I will try to dig deeper.

larstobi, 2020-12-03:
Same symptoms with a ZW111 Nano Dimmer here.

staal0, 2020-12-16:
Same with Logicgroup ZDB5100 and OpenZWave 1.6.1545.

MarkGodwin, 2020-12-27:
Same with the ZMNHSDx DIN Dimmer. Switch states and power levels aren't updated when they change. If I re-include the device the current state and power level comes through, but then no further updates after that.

Many hours searching the forums, but I have no clue what to do to fix it.

I'm trying to move from Hubitat - I had other issues with that, but the power level and switch state reporting did work.

hogend, 2021-01-26:
Same problem with a Fibaro FGS223 Double Relay Switch and OZWDaemon 0.1.182 (OpenZWave 1.6.1570).
The problem is aggravated by the fact that openzwave seems to proxy the state of the switch. If the proxy state is "on", new "on" commands are not transmitted to the device. So, since it is, in reality, "off", it stays "off".

MCLW, 2021-01-26:
Same problem with Fibaro Roller Shutter 3 FGR-223.