OpenZWave/open-zwave #2455: Aeotec Wallmote Quad (Central Scene) no longer working after OZW upgrade

tommyjlong, 2020-11-09:
Context - I use ZWave2MQTT.  When I upgrade ZWave2MQTT, I upgrade to its latest tagged version, and at the same time I also upgrade OZW to the master branch.

Problem - After upgrading ZWave2MQTT to v4.0.5 and upgrading OZW to 1.6.1392, the Aeotec Wallmote Quad (Central Scene) quit working.

I reinstalled the shared libraries of OZW back down to 1.6.1142, but kept ZWave2 MQTT at v4.0.5, and the Wallmote Quad works again.

Here are a couple of logs:
**Case NOT WORKING - OZW 1.6.1392, Z2m v4.0.5. Press Button 1 momentarily**
```
Nov 09 17:02:00 NUC node[17487]: m2020-11-09 17:02:00.127 Detail, Node008,   Received: 0x01, 0x0f, 0x00, 0x04, 0x00, 0x08, 0x09, 0x60, 0x0d, 0x00, 0x01, 0x5b, 0x03, 0x3f, 0x00, 0x01, 0xff
Nov 09 17:02:00 NUC node[17487]: 2020-11-09 17:02:00.127 Detail,
Nov 09 17:02:00 NUC node[17487]: 2020-11-09 17:02:00.127 Info, Node008, MultiChannelEncap with endpoint set to 0 - Send to Root Device
Nov 09 17:02:00 NUC node[17487]: 2020-11-09 17:02:00.127 Info, Node008, Received Central Scene set from node 8: scene id=1 with key Attribute 0. Sending event notification.
Nov 09 17:02:00 NUC node[17487]: 2020-11-09 17:02:00.127 Warning, Node008, No ValueID created for Scene 1
Nov 09 17:02:06 NUC node[17487]: 2020-11-09 17:02:06.057 Detail, Node008,   Received: 0x01, 0x0d, 0x00, 0x04, 0x00, 0x08, 0x07, 0x60, 0x0d, 0x01, 0x01, 0x20, 0x01, 0x00, 0xb5
Nov 09 17:02:06 NUC node[17487]: 2020-11-09 17:02:06.058 Detail,
Nov 09 17:02:06 NUC node[17487]: 2020-11-09 17:02:06.060 Info, Node008, Received a MultiChannelEncap from node 8, endpoint 1 for Command Class COMMAND_CLASS_BASIC
Nov 09 17:02:06 NUC node[17487]: 2020-11-09 17:02:06.061 Info, Node008, Received Basic set from node 8: level=0.  Sending event notification.
Nov 09 17:02:06 NUC node[17487]: 2020-11-09 17:02:06.062 Detail, Node008, Notification: NodeEvent
Nov 09 17:02:06 NUC node[17487]: [53B blob data]
Nov 09 17:02:06 NUC node[17487]: m2020-11-09 17:02:06.395 Detail, Node008,   Received: 0x01, 0x0d, 0x00, 0x04, 0x00, 0x08, 0x07, 0x60, 0x0d, 0x01, 0x01, 0x20, 0x01, 0x00, 0xb5
Nov 09 17:02:06 NUC node[17487]: 2020-11-09 17:02:06.396 Detail,
Nov 09 17:02:06 NUC node[17487]: 2020-11-09 17:02:06.396 Info, Node008, Received a MultiChannelEncap from node 8, endpoint 1 for Command Class COMMAND_CLASS_BASIC
Nov 09 17:02:06 NUC node[17487]: 2020-11-09 17:02:06.396 Info, Node008, Received Basic set from node 8: level=0.  Sending event notification.
Nov 09 17:02:06 NUC node[17487]: 2020-11-09 17:02:06.396 Detail, Node008, Notification: NodeEvent
Nov 09 17:02:06 NUC node[17487]: [53B blob data]
Nov 09 17:02:07 NUC node[17487]: m2020-11-09 17:02:07.110 Detail, Node008,   Received: 0x01, 0x0e, 0x00, 0x04, 0x00, 0x08, 0x08, 0x60, 0x0d, 0x01, 0x01, 0x26, 0x01, 0x00, 0xff, 0x40
Nov 09 17:02:07 NUC node[17487]: 2020-11-09 17:02:07.111 Detail,
Nov 09 17:02:07 NUC node[17487]: 2020-11-09 17:02:07.111 Info, Node008, Received a MultiChannelEncap from node 8, endpoint 1 for Command Class COMMAND_CLASS_SWITCH_MULTILEVEL
Nov 09 17:02:07 NUC node[17487]: 2020-11-09 17:02:07.112 Warning, Node008, Routing HandleIncomingMsg to HandleMsg - Please Report: COMMAND_CLASS_SWITCH_MULTILEVEL
Nov 09 17:02:07 NUC node[17487]: 2020-11-09 17:02:07.112 Warning, Node008, Recieved a Unhandled SwitchMultiLevel Command: 1
Nov 09 17:02:07 NUC node[17487]: 2020-11-09 17:02:07.113 Warning, Node008, MultiChannel Encap CommandClass COMMAND_CLASS_SWITCH_MULTILEVEL HandleIncomingMsg returned false
Nov 09 17:02:07 NUC node[17487]: 2020-11-09 17:02:07.472 Detail, Node008,   Received: 0x01, 0x0e, 0x00, 0x04, 0x00, 0x08, 0x08, 0x60
```

**Case WORKING: OZW 1.6.1142, Z2m v4.0.5.  Press Button 1 momentarily**
```
Nov 09 17:20:36 NUC node[18134]: 2020-11-09 17:20:36.154 Detail, Node008,   Received: 0x01, 0x0f, 0x00, 0x04, 0x00, 0x08, 0x09, 0x60, 0x0d, 0x00, 0x01, 0x5b, 0x03, 0x42, 0x00, 0x01, 0x82
Nov 09 17:20:36 NUC node[18134]: 2020-11-09 17:20:36.155 Detail,
Nov 09 17:20:36 NUC node[18134]: 2020-11-09 17:20:36.155 Info, Node008, MultiChannelEncap with endpoint set to 0 - Send to Root Device
Nov 09 17:20:36 NUC node[18134]: 2020-11-09 17:20:36.155 Info, Node008, Received Central Scene set from node 8: scene id=1 with key Attribute 0. Sending event notification.
Nov 09 17:20:36 NUC node[18134]: 2020-11-09 17:20:36.156 Detail, Node008, Initial read of value
Nov 09 17:20:36 NUC node[18134]: 2020-11-09 17:20:36.158 Info, Node008, Automatically Clearing Scene 1 in 1000ms
Nov 09 17:20:36 NUC node[18134]: 2020-11-09 17:20:36.158 Info, Timer: adding event in 1000 ms
Nov 09 17:20:36 NUC node[18134]: 2020-11-09 17:20:36.159 Detail, Node008, Notification: ValueChanged CC: COMMAND_CLASS_CENTRAL_SCENE Instance: 1 Index: 1
Nov 09 17:20:36 NUC node[18134]: 2020-11-09 17:20:36.160 Detail, Timer: waiting with timeout 999 ms
Nov 09 17:20:36 NUC node[18134]: [115B blob data]
Nov 09 17:20:37 NUC node[18134]: m2020-11-09 17:20:37.160 Info, Timer: delayed event
Nov 09 17:20:37 NUC node[18134]: 2020-11-09 17:20:37.160 Detail, Node008, Refreshed Value: old value=1, new value=0, type=list
Nov 09 17:20:37 NUC node[18134]: 2020-11-09 17:20:37.160 Detail, Node008, Changes to this value are not verified
Nov 09 17:20:37 NUC node[18134]: 2020-11-09 17:20:37.161 Detail, Timer: waiting with timeout -1 ms
Nov 09 17:20:37 NUC node[18134]: 2020-11-09 17:20:37.161 Detail, Node008, Notification: ValueChanged CC: COMMAND_CLASS_CENTRAL_SCENE Instance: 1 Index: 1
Nov 09 17:20:37 NUC node[18134]: [115B blob data]
Nov 09 17:20:37 NUC node[18134]: m2020-11-09 17:20:37.373 Detail, Node008,   Received: 0x01, 0x0f, 0x00, 0x04, 0x00, 0x08, 0x09, 0x60, 0x0d, 0x00, 0x01, 0x5b, 0x03, 0x42, 0x00, 0x01, 0x82
Nov 09 17:20:37 NUC node[18134]: 2020-11-09 17:20:37.373 Detail,
Nov 09 17:20:37 NUC node[18134]: 2020-11-09 17:20:37.373 Info, Node008, MultiChannelEncap with endpoint set to 0 - Send to Root Device
Nov 09 17:20:37 NUC node[18134]: 2020-11-09 17:20:37.373 Warning, Node008, Received Duplicated Scene Notification. Dropping...
Nov 09 17:20:37 NUC node[18134]: 2020-11-09 17:20:37.552 Detail, Node008,   Received: 0x01, 0x0f, 0x00, 0x04, 0x00, 0x08, 0x09, 0x60, 0x0d, 0x00, 0x01, 0x5b, 0x03, 0x42, 0x00, 0x01, 0x82
Nov 09 17:20:37 NUC node[18134]: 2020-11-09 17:20:37.553 Detail,
Nov 09 17:20:37 NUC node[18134]: 2020-11-09 17:20:37.555 Info, Node008, MultiChannelEncap with endpoint set to 0 - Send to Root Device
Nov 09 17:20:37 NUC node[18134]: 2020-11-09 17:20:37.556 Warning, Node008, Received Duplicated Scene Notification. Dropping...
Nov 09 17:20:39 NUC node[18134]: 2020-11-09 17:20:39.020 Detail, Node008,   Received: 0x01, 0x0f, 0x00, 0x04, 0x00, 0x08, 0x09, 0x60, 0x0d, 0x00, 0x01, 0x5b, 0x03, 0x42, 0x00, 0x01, 0x82
Nov 09 17:20:39 NUC node[18134]: 2020-11-09 17:20:39.020 Detail,
Nov 09 17:20:39 NUC node[18134]: 2020-11-09 17:20:39.021 Info, Node008, MultiChannelEncap with endpoint set to 0 - Send to Root Device
Nov 09 17:20:39 NUC node[18134]: 2020-11-09 17:20:39.021 Warning, Node008, Received Duplicated Scene Notification. Dropping...

```

kpine, 2020-11-10:
The cache version changed between those OZW versions, and when it upgrades it triggers a complete wipe of the cachefile. If the Wallmote did not refresh on its own (is it battery powered?) then you need to manually refresh it to create the OZW values that represent the scenes.

The message you are seeing here occurs when the values are missing from the cache. A refresh will repopulate them.
```
Nov 09 17:02:00 NUC node[17487]: 2020-11-09 17:02:00.127 Info, Node008, Received Central Scene set from node 8: scene id=1 with key Attribute 0. Sending event notification.
Nov 09 17:02:00 NUC node[17487]: 2020-11-09 17:02:00.127 Warning, Node008, No ValueID created for Scene 1
```

tommyjlong, 2020-11-10:
Thanks for getting back. Yes the Wallmote is battery powered.  I had actually done that already...but to make sure I'm understanding correctly here is what I did:
After Z2m restart, it found all the other main powered devices, I did awake the Wallmote (hold its button for 3 secs) and while viewing the OZW logs I could see the exchange of information with the wallmote and OZW; and subsequently Z2m did populate its entry for the wallmote.  I did not capture the logs and do no know if the exchange went to "Complete". The wallmote then fell back to sleep.  I next pressed the button and the above are the logs I took. 

I have done this procedure for upgrades and downgrades and have repeated this for both upgrade/downgrade cycles a couple of times. Only 1.6.1142 works.  I will try a few more times and report back.

tommyjlong, 2020-11-10:
I've got it working now.  But I want to report that of my two battery operated devices, it was very difficult getting all of their parameters discovered.  I would wake it up, and it would like discover some of the non-config class parameters, but not the manufacturer id and type.  Wake it up several more times to no avail.  Restart z2m and wake the device and it would discover the manufacturer id and type but not all of the non-config class parameters.  It is very hit and miss.  Anyway you can close this.

RicardP, 2020-11-11:
This "half complete" interview is a big problem for me as well, it is an old kown flaw by the Node FW, Z-Wave or I am guessing OZW not handling the interview process correctly if battery devices?

tommyjlong, 2020-11-12:
As things are working for now, I will close this.

closed 2020-11-12 by tommyjlong