OpenZWave/open-zwave #2461: Collaboration on devices DB with node-zwave-js

robertsLando, 2020-11-12:
Hi @Fishwaldo, I would like to contribute configuration files to another zwave related project named [node-zwave-js](https://github.com/zwave-js/node-zwave-js). I’ve experimented with importing and translating openzwave devices configuration xml files into the format required by node-zwave-js.

That library has really strict configuration checks in place and during my experiments I have found many errors in the OZW configuration files, like min/max/default values that are incompatible with the value size, duplicate config parameter labels, missing options for list-type configuration params, etc. [this](https://github.com/OpenZWave/open-zwave/pull/2460) PR is an example in the manufacturers file.

I believe that collaborating on the config files would greatly benefit both projects. I’ve talked to the maintainer @AlCalzone and he is okay with importing the OZW configuration files if the fixes are contributed back to OZW and if you are okay with it.

What do you think about this?

jhelmink, 2021-01-11:
Based on @Fishwaldo reply here; https://community.home-assistant.io/t/can-we-revisit-the-move-to-qt-openzwave/262612/140?u=tlf
I think it's unlikely that you'll get a response for a while.

IMHO the best you can do is just create PR's with the fixes, and one day they might eventually get approved.

robertsLando, 2021-01-11:
@jhelmink PR's with the fixes on Openzwave or on zwave-js?

jhelmink, 2021-01-11:
@robertsLando I was thinking on Openzwave, zwave-js will need to be the new masters if qt-openzwave is no longer being maintained, but I just wanted to point out that you may not get a response to this issue.