OpenZWave/open-zwave #2462: Zooz ZEN34

broyuken, 2020-11-15:
Zooz came out with a new remote switch that is battery powered. Please add support in open zwave for it.

https://www.thesmartesthouse.com/collections/light-switches/products/zooz-z-wave-plus-700-series-remote-switch-zen34-battery-powered

kpine, 2020-11-15:
Can you be more specific, what is not working?

broyuken, 2020-11-15:
It shows up as an unknown device because it's not in the openzwave database. Extremely similar request to https://github.com/OpenZWave/open-zwave/issues/2119 but just with a different new device.

kpine, 2020-11-15:
The database is community maintained. Your best bet is to create the XML file yourself and submit a PR. https://github.com/OpenZWave/open-zwave/wiki/Adding-Devices

hoopsta1423, 2020-11-16:
> It shows up as an unknown device because it's not in the openzwave database. Extremely similar request to #2119 but just with a different new device.

I took a stab at this and got it working

add this to the manufacturer_specific.xml under Manufacturer id="027a"

    <Product config="zooz/zen34.xml" id="F001" name="ZEN34 Remote Switch" type="0004"/>

Then unzip this file to the config/zooz directory:
[zen34.zip](https://github.com/OpenZWave/open-zwave/files/5547026/zen34.zip)

broyuken, 2020-11-16:
I did this but I'm still not getting a switch. Only battery and a couple of disabled sensors. No switch entity. Enabling the other sensors didn't seem to help. They don't change when I hit the paddle up or down.

hoopsta1423, 2020-11-16:
as far as I'm aware, that is how this device is supposed to work. The sensors you refer to are scenes for the upper and lower paddles (click 1-5 times, paddle held, paddle released). You would use those scenes to, say, toggle a light/outlet when activated.

It also supports associations so you can pair it directly to another zwave device for single tap on/off, or dimming

hoopsta1423, 2020-11-16:
> I did this but I'm still not getting a switch. Only battery and a couple of disabled sensors. No switch entity. Enabling the other sensors didn't seem to help. They don't change when I hit the paddle up or down.

as far as I'm aware, that is how this device is supposed to work. The sensors you refer to are scenes for the upper and lower paddles (click 1-5 times, paddle held, paddle released). You would use those scenes to, say, toggle a light/outlet when activated.

It also supports associations so you can pair it directly to another zwave device for single tap on/off, or dimming

broyuken, 2020-11-16:
The sensors I get in Home Assistant are:
Zen34 Remote Switch: Basic
Zen34 Remote Switch: Basic Duration
Zen34 Remote Switch: Basic Target
Zen34 Remote Switch: Batter Level

The only one that seems to change at all is the battery level when I press up or down on the paddle. For my GE switches for example I get a switch entity that turns off and on like a normal switch when I do a single tap, and a Basic entity which goes to 255 if I double tap on and 0 if I double tap off. With this, none of the sensors change.

hoopsta1423, 2020-11-16:
What are you using for zwave in HA? I use zwave2mqtt and I get the proper entities

battery level
scene state 1 (lower paddle)
scene state 2 (upper paddle)

Did you exclude and re-include after updating manufacturer_specific and adding zen34.xml?

broyuken, 2020-11-16:
I'm using the beta openzwave integration. I moved off of zwave2mqtt a few months back because I couldn't use transitions.

hoopsta1423, 2020-11-16:
> I'm using the beta openzwave integration. I moved off of zwave2mqtt a few months back because I couldn't use transitions.

Hmm, I'm not sure then. Kind of unrelated but I was having issues with OZWbeta with the ZEN31 RGBW dimmer so I moved that to zwave2mqtt as well and it works perfectly.

broyuken, 2020-11-16:
Thanks for the info! I have a support ticket open with zooz on this as well. If I find something that works for ozw beta I'll be sure to post it here for others as well.

broyuken, 2020-11-19:
Could you send me a screenshot of what your sensors look like, and what state they go into when you activate a scene?

hoopsta1423, 2020-11-19:
[image]

I pressed the upper paddle once so you'll see that the state for scene_state_1 is "Pressed 1 time" . It resets back to "Inactive" after 1000ms. You can also see the values (press 1-5 times, key held, key released)

scene_state_1 is upper paddle
scene_state_2 is lower paddle

There's also a light entity which I assume works with associations? I haven't played with that just yet though

There's another sensor created by zwave2mqtt that I remove because its just a count of scenes

broyuken, 2020-11-19:
[image]
This is all I get, I don't get the scene state entities.

hoopsta1423, 2020-11-19:
Yea, I don't know. Must be something with ozwbeta

kpine, 2020-11-19:
There's not such thing as "scene state entities" in the HA OpenZWave beta. You use [events](https://www.home-assistant.io/integrations/ozw/#event-ozwscene_activated).

broyuken, 2020-11-19:
So I would have to listen to events and create template entities?

kpine, 2020-11-19:
If you want to create a template entity, that's entirely up to you.

broyuken, 2020-11-19:
In case anyone else needs this in the future, here's what I did to get it working.

Trigger:
```
platform: event
event_type: ozw.scene_activated
event_data:
  node_id: 21
```

Condition:
```
condition: not
conditions:
  - condition: template
    value_template: trigger.event.data.scene_value_label == 'Inactive'
```

Action:
```
service: >-
  {% if (trigger.event.data.scene_label == 'Scene 1') and
  (trigger.event.data.scene_value_label == 'Pressed 1 Time') %}light.turn_on{%
  elif (trigger.event.data.scene_label == 'Scene 2') and
  (trigger.event.data.scene_value_label == 'Pressed 1 Time')%}light.turn_off{%
  endif %}
data:
  entity_id: light.playroom_light
```

commit aa342655 (2021-01-03): Bumped close #2462
closed 2021-01-03 by nechry