OpenZWave/open-zwave #2475: Something in nas-pd03z.xml makes openzwave daemon crash

wauswaus, 2020-11-21:
Hi,

I have a Shenzen-Neo PD03Z for which I added a [device specific xml file](https://github.com/OpenZWave/open-zwave/blob/master/config/shenzen_neo/nas-pd03z.xml) . I'm running the openzwave beta in Home Assistant but they made it impossible to get access to all the config files so I could not test this file and I made a pull request, keeping my fingers crossed it would work. Unfortunately not... If I add my device the openzwave daemon crashes. See the attached log.

I want to fix this but I'm having some trouble getting to the cause of this so I need your help! Log is attached.
I copied the file of the PD02Z and made minor modifications. 
I think I fixed:
```
[20201121 17:05:18.547 CET] [ozw.library] [warning]: Warning - Node: 0 Attempt to get a Invalid Index 12 on ValueList HomeID: 0xfd807f47, ValueID: (Id 0x00010000159c0014, NodeID 21, Genre config, CC 0x70, Instance 1, Index 1, Type list) 
[20201121 17:05:18.548 CET] [ozw.library] [warning]: Warning - Node: 0 Value is not found in xml configuration for node 21, class 0x70, instance 1, index 1 - HomeID: 0xfd807f47, ValueID: (Id 0x00010000159c0014, NodeID 21, Genre config, CC 0x70, Instance 1, Index 1, Type list) 
```
There was a list defined but no list items for index 1. 

Leaves 
````
[warning]: Warning - Node: 0 Attempt to get a Invalid Index 1 on ValueList HomeID: 0xfd807f47, ValueID: (Id 0x00040000159c0014, NodeID 21, Genre config, CC 0x70, Instance 1, Index 4, Type list) 
[20201121 17:05:18.549 CET] [ozw.library] [warning]: Warning - Node: 0 Value is not found in xml configuration for node 21, class 0x70, instance 1, index 4 - HomeID: 0xfd807f47, ValueID: (Id 0x00040000159c0014, NodeID 21, Genre config, CC 0x70, Instance 1, Index 4, Type list) 
````
I think this is because the default value of 1 does not exist in any of the list items.

And of course the crashing.

I hope you can help!
[log.txt](https://github.com/OpenZWave/open-zwave/files/5578111/log.txt)

l3arcf, 2020-11-23:
@wauswaus , in lieu of other responses I tool a look at index 1 of command-class 112 that you have in your config and see it's a list, but you have not defined the list options.  Maybe that's the issue?  I have a PD02Z and that config has the following options:

      <Item label="High Sensitivity" value="8" />
      <Item label="Normal Sensitivity" value="12" />
      <Item label="Lower Sensitivity" value="128" />
      <Item label="Lowest Sensitivity" value="255" />

I wonder if you need something like that.

wauswaus, 2020-11-23:
Made a [pull request](https://github.com/OpenZWave/open-zwave/pull/2480), so lets hope for the best.

closed 2020-11-23 by wauswaus