OpenZWave/open-zwave #2481: DSB09104 wrong value return for parameter 1

chatainsim, 2020-11-24:
Hello,
I have a DSB09104 energy meter from Aeotec.
The first parameter is the voltage and openzwave always display it at 120 instead of 240.

I've contacted the Aeotec support and give me some hit.

Using Zensys Tools, I was able to change the value to 240.
Using this tool to get parameter 1 value:

```
04:11:12.256: Completed: 030 - COMMAND_CLASS_CONFIGURATION.CONFIGURATION_GET, data: 01
04:11:12.308: Received: 030 - COMMAND_CLASS_CONFIGURATION.CONFIGURATION_REPORT, data: 01 02 00 F0 00 00
```
As you can see it's F0, so 240.
But in openzwave control panel, it's always 120:
[image]

Do you know how to fix this ?
Thank you.
Regards,

nechry, 2021-01-03:
did you refresh configuration parameter after you change it from zensys tool?

chatainsim, 2021-01-03:
Yes, I have refresh configuration.

nechry, 2021-01-03:
ok and if you try to set again from ozw you still get 1 ? I don't have anymore this device to test.

chatainsim, 2021-01-03:
Yes, the parameters is not read correctly. Even if it's set to 240 ozw still display 120

nechry, 2021-01-04:
is a old device and I can't get the engineering sheet. can be parameter size mistake. 
anyway if you change the parameter with zensys tool and you use the device on ozw the value you set is save on the device memory. the power calculation will be good

chatainsim, 2021-01-07:
Ok, thank you for the help.

closed 2021-01-07 by chatainsim