OpenZWave/open-zwave #2489: Shenzen Neo Coolcam Power Plug new version

Huppeltje16, 2020-12-02:
Device: Neo Coolcam Power Plug 12A+

After purchasing some Neo Coolcam Power Plugs, Domoticz eg. OZW did not recognize them.
By just adding the lines below to the manufacturer_specific.xml, everything works fine. The new plugs only have different ID's and type, everything else is the same as for the previous version.
------------------------------------------------------------------------------------------------
  <Manufacturer id="041a" name="Shenzhen Neo Electronics Co Ltd">
    <Product config="shenzen_neo/nas-wr01ze.xml" id="0008" name="NAS-WR01ZE Power plug 12A" type="0200"/>
  </Manufacturer>
------------------------------------------------------------------------------------------------

Kind regards,
Petro Boef

ttjorvi, 2020-12-17:
How did they mess up your network? I have a couple that I need to get working and add to the manufacturer...xml file. So it would be good to know what is wrong.

Huppeltje16, 2020-12-28:
Hi,

I don’t know how, but the Zwave network repeatedly crashed while these plugs were connected. Also when I only connected one. I have no possibility to trace the Zwave network, so I consider these plugs to be worthless now.

Petro

De informatie in dit bericht is uitsluitend bestemd voor de geadresseerde. Aan dit bericht en de bijlagen kunnen geen rechten worden ontleend. Heeft u deze e-mail onbedoeld ontvangen? Dan verzoeken wij u het te vernietigen en de afzender te informeren. Openbaar maken, kopiëren en verspreiden van deze e-mail of informatie uit deze e-mail is alleen toegestaan met voorafgaande schriftelijke toestemming van de afzender. Het Radboudumc staat geregistreerd bij de Kamer van Koophandel in het handelsregister onder nummer 41055629.

The content of this message is intended solely for the addressee. No rights can be derived from this message or its attachments. If you are not the intended recipient, we kindly request you to delete the message and inform the sender. It is strictly prohibited to disclose, copy or distribute this email or the information inside it, without a written consent from the sender. Radboud university medical center is registered with the Dutch Chamber of Commerce trade register with number 41055629.