OpenZWave/open-zwave #2520: manufacturer_specific.xml - type and id probably inverted for MHS314 (MCO Home) device

loomyr, 2020-12-29:
Hi, 
When adding a MCO-Home MH-S314 switch, OZW was unable to detect it correctly : 

[image]

After checking the "**_manufacturer_specific.xml_**" file I figured out that the type and the id of the MCO-Home MH-S314 might be inverted: 
https://github.com/OpenZWave/open-zwave/blob/776eeda3b571b5f82c77166a60878fa75eb5c499/config/manufacturer_specific.xml#L1229

Based on my device inclusion (attached picture above), I assume it should be referenced like that :

> `id="5102" name="MH-S314 Four-load" type="3141"`

Best Regards,
Vincent

Hystersis, 2020-12-30:
I found the same inversion when updating the MH-S513 (see PR #2516).

stuartwishart, 2020-12-31:
I found a similar issue with the MCO MH-S513 definition, see PR #2516

commit 825f10c9 (2021-01-03): fix #2520  type and id inverted for MHS314
closed 2021-01-03 by nechry