OpenZWave/open-zwave #2529: Recent changes to Aeotec switch broke state updates

abmantis, 2021-01-05:
The following pull request broke the state reporting of the relays of Aeotec Switches: https://github.com/OpenZWave/open-zwave/pull/2517

Now, when pressing the physical switch, on the log I get "No Valid Mapping for Basic Command Class and No ValueID Exported. Error?", and the state of the device does not change.
State report was working fine before.

matejdro, 2021-01-06:
Do you use basic command class to receive state updates (e.g. do you have your Z-wave controller added to the groups of your Aoetec switch)?

kpine, 2021-01-06:
If I'm guessing, this is a Nano Switch? The best option for something like a Nano Switch is to set parameter 80 to "Basic". The switch sends 0/255 to signal off/on, and OZW does a single GET to refresh the binary switch value. Another choice would be to set 80 to "Hail". It would probably workaround the issue, but could cause more network traffic because OZW will do more GETs.

matejdro, 2021-01-06:
I guess good workaround would be to set it to `Hail CC when using the external switch to switch the loads`. That will only cause more network traffic when you use physical switch, but not when you toggle it via OZW (which sends get after set anyway so it shouldn't stay out of sync).

kpine, 2021-01-06:
Are you saying these Aeotec switches also send Reports/Hail when you toggle it with a z-wave command? Then in that case, "Hail CC" would be a bad choice, and the "external switch" version is what I had in mind.

matejdro, 2021-01-06:
Yup that's exactly what it's happening.

abmantis, 2021-01-06:
> Do you use basic command class to receive state updates (e.g. do you have your Z-wave controller added to the groups of your Aoetec switch)?

I'm not sure (still new to ZWave), but I think Zwave2Mqtt adds itself to group 0, so yes?

> I guess good workaround would be to set it to `Hail CC when using the external switch to switch the loads`. That will only cause more network traffic when you use physical switch, but not when you toggle it via OZW (which sends get after set anyway so it shouldn't stay out of sync).

I've tried this and it solves the issue for the external switch. Unfortunately, it still fails when toggling it view OZW, because it sends get right after set, and the module replies with the old state like 80% of the times. This is because by the time the module replies to GET, it still hasn't updated the state yet. So the reply to get is "oldState = false, newState = false". 
fishwaldo explains the issue here for dimmers, but its the same problem: see https://community.home-assistant.io/t/dimmer-lights-show-as-on-even-when-turned-off/152975/10 
Having the basic report would always keep the state in sync.

If there's a solution to the Get after Set issues, I would be fine having Hail CC for external switches.

matejdro, 2021-01-06:
> I'm not sure (still new to ZWave), but I think Zwave2Mqtt adds itself to group 0, so yes?

Yeah sorry, I mixed it up with Nano Shutter which sends Multilevel switch updates to group 0

Huh that's weird, I do not experience that. Get after set is usually pretty reliable for me.

abmantis, 2021-01-06:
> Huh that's weird, I do not experience that. Get after set is usually pretty reliable for me.
On Nano Switches? Even after the changes disabling the basic report mapping?
For me, most of the times the state reported by get is the old one. I think they have the latest FW.

nechry, 2021-01-06:
I can rollback the PR. I don't have time to test with devices.

closed 2021-01-06 by nechry

matejdro, 2021-01-06:
Completely understandable. Thanks for your help.

PR #2530 (2021-01-06): Revert "Add basic command class to Aeotec switches and dimmers"
Reverts OpenZWave/open-zwave#2517
Close #2529