OpenZWave/open-zwave #2556: Qubino ZMNHSDx DIN Dimmer Lifeline

MarkGodwin, 2021-01-25:
I've been experiencing issues with the inclusion of the above device. I can control the device correctly, but updates from the device such as power levels and changes due to manual switching do not get reported back to the controller.

This is with a ZMNHSDx (Qubino DIN Dimmer), and an AEon Z-Stick Gen5+.

The problem seems to be down to the lifeline association. I have confirmed this with the Silabs PC Controller sample program.

Using version 1.4.3469 of OpenZWave from the old HomeAssistant integration, the lifeline for the devices is set to "1" when they are included, and updates to the switch and power states get propagated correctly.

Using version 1.6.1710 from the OpenZWave (ozwdaemon) beta integration in HA, the lifeline for the devices is set to "1.1", and the updates of switch state changes from the device do not work.

The only way I know to verify this is with the PC Controller app. It seems that if I manually update the lifeline group to "1" in the PC Controller, the problem is fixed.

I also note that if I include the devices via the PC Controller app (version 5.50.102), the lifeline is set to 1, and then moving the Z-stick back to HA/OpenZWave also works.

There are a few other issue reports concerning the 1 vs 1.1 problem with OpenZWave, but I have to admit that most of the discussion went right over my head. I'm not sure where the fault lies. There does not seem to be any way to get the OZW_Log.txt from ozwdaemon, though.

depetewes, 2021-02-03:
I have exact same issue, controller ZW090 Z-Stick Gen5 EU (AEON Labs) and Qubino DIN dimmer ZMNHSDx. Nu updates when operated manually/direct. I experienced this when updateing Z-Wave to MQTT to version 0.9.0.

MarkGodwin, 2021-02-03:
If it helps, zwave-js (the new integration in Home Assistant 2021.2.0 beta) does not seem to have the same problem.