OpenZWave/open-zwave #2563: Can we get the Zooz Zen77 switches added to OZW?

peterbaker, 2021-01-28:
Here are the parameters and values for the switch.
https://www.support.getzooz.com/kb/article/550-zen77-s2-dimmer-700-advanced-settings/

nechry, 2021-01-28:
@peterbaker did you take a look on the Wiki https://github.com/OpenZWave/open-zwave/wiki/Adding-Devices for adding a new device?
If you can create the xml file this will be easy.

peterbaker, 2021-01-28:
@nechry Ah great, I hadn't seen that. I'll give it a try.

deiger, 2021-02-01:
@peterbaker please note that this is a Z-Wave 700 switch, so it will be supportable only in the next version of OpenZWave (v1.7).

peterbaker, 2021-02-01:
> @peterbaker please note that this is a Z-Wave 700 switch, so it will be supportable only in the next version of OpenZWave (v1.7).

Oh crap, for real? It feels like I've been hearing about 1.6 becoming adopted "soon" for a couple of years, and now 1.7 is needed for a whole new set of stuff coming out? Any expectation that 1.7 will be incorporated into systems like Home Assistant any quicker than 1.6 was/is?

deiger, 2021-02-01:
Now that HA is integrated with OpenZWave through an add-on, it should be fairly quickly, once 1.7 is rolled out. When that will be? #2570

cybergrimes, 2021-02-02:
700 series devices should work just fine on OZW 1.6, I've got the Zooz ZEN34 remote switch working on 1.4 even

blhoward2, 2021-02-16:
> Now that HA is integrated with OpenZWave through an add-on, it should be fairly quickly, once 1.7 is rolled out. When that will be? #2570

Uh, about that. The maintainer of ozw said he was taking a step back so HA changed its development path to focus on node-zwave-js.

cybergrimes, 2021-02-16:
Zooz ZEN77 added in https://github.com/OpenZWave/open-zwave/pull/2581

The same config file can be edited to use with OZW 1.4 if necessary

closed 2021-02-17 by nechry