OpenZWave/open-zwave #2570: Timeline for v1.7

deiger, 2021-02-01:
Thank you for all your investment in this project.
I understand that support for Z-Wave 700 will only be added in [v1.7](../milestone/3).
Is there any update on the timeline for it?

JGR-66, 2021-02-01:
I suggest you move your hopes to "OZWay", it is based on ZWay.me with an OZW wrapper, many users are frustrated over OZW. ZWay is an official ZWave software, still developed by enthusiasts, and it seems follow ZWave evolution nicely. I hope it gets released soon :)

deiger, 2021-02-02:
If [this](https://github.com/Z-Wave-Me/ozway) is what you are referring to, the fork still has no internal updates. I'm also not sure relying on yet another 3rd party that sells its software is my best bet.

Honestly, I'm surprised Silicon Labs are not sponsoring and heavily contributing to OpenZWave.  At the current atmosphere, with [Connected Home over IP](https://www.connectedhomeip.com/) sponsored by Google, Amazon, Apple, Samsung and the Zigbee Alliance, it might be the only thing standing between Z-Wave and oblivion.

JGR-66, 2021-02-02:
I read OZW project owner says he cannot engage himself any longer, that is okay of course his choice, but he says also no one else is interested in taking over his role or even team up with him (as I understood it). For me that sounds very unlikely as so many skilled programmers work on both Hass and Domoticz, just to mention. And I never seen a call out for help here for 4 years, I checkout everything here at least once a week. 
I would gladly stay with OZW and contribute with at least user feedback and donations but motivation for both feels low and meaningless - since a long time... People makes Pull Requests and posts Issues and solutions but the situation here leaves everyone in limbo.

blhoward2, 2021-02-16:
Take a look at node-zwave-js (and zwave2jsmqtt). We've already got some support for the 700 sticks and it's being very actively developed.

deiger, 2021-02-16:
> Take a look at node-zwave-js (and zwave2jsmqtt). We've already got some support for the 700 sticks and it's being very actively developed.

Already switched, thanks.