OpenZWave/open-zwave #2572: Open zwave error is causing spike in total kWh reading

GravityRZ, 2021-02-03:
Domoticz version 2020.2.12194
Open Zwave Version: 1.6-1185-g0dba812c
Qubino ZMNHADX  module address 11

sometimes(1 in  a couple of months) i am getting this error

`2021-02-02 21:30:26.507 Error: OpenZWave: Value_Changed: Tried adding value, not succeeded!. Node: 11 (0x0b), CommandClass: SENSOR MULTILEVEL, Label: Time, Instance: 1, Index: 33`

not a big deal because  everything on that module seems to work ok, also after the error has occured

I Also got rarely a kWh spike on that same devices kWh meter. also not a big deal because al i had to do is delete the faulty spike reading and everything was normal again.
The spike only appears in the kWh device, the Watt device is normal

Today i found out that that error is causing the spike. somehow the error screws up the kWh reading and sends out a very high amount.
The spike amount looks like the total of 1 months

Is this a known bug?

[image]

JGR-66, 2021-02-05:
It is more likely caused by Domoticz, just after reboot. Did you reboot prior this reading? I have the same issue since few months, something was done with the code for Energy measure/graphs. I recommend you to post this to Domoticz developer issues instead :)

GravityRZ, 2021-02-05:
no it was running stable for months. i will see if i  will post this on the domoticz github

closed 2021-02-05 by GravityRZ

JGR-66, 2021-02-06:
If it is happening from time to time, not just after reboot, it is likely a ZWave/OZW issue. And random corrupt values is most likely caused by an interfered radio packet that by chance after demodulation of data had a correct CRC check, and therefore accepted by the Controller.
In my experience the worse RF link quality to a Node the more often you get corrupt measures from OZW. But of course also a buggy Node Firmware could be the reason, however I don't believe it is the problem in your case as it happens so rarely.