OpenZWave/open-zwave #2573: Problem with FGS-224 S2 state

03397, 2021-02-04:
### Issue Guidelines

Before opening a new issue, please review the following:
* The Github Issues is for OZW Bugs only. Please do not use this for general questions or howto type conversations. If you have a question, please post to our mailing list at https://groups.google.com/forum/#!forum/openzwave
* Please use the search feature to see if your issues have been raised or addressed first.
* Try the OpenZWave Log Analyzer at http://www.openzwave.com/log-analyzer if you are experiencing problems with a device on your network.
* Often we will require Log Files to help diagnose the issue. Please stop any application you are using, remove zwcfg_*.xml/ozwcache_*.xml files, and restart your application. After experiencing your issue, stop your application and attach the OZW_Log.txt file to this issue (please do not filter the log file, we often need the full log file to diagnose issues)
* If you want to add a new device to the OZW Database, please consult https://github.com/OpenZWave/open-zwave/wiki/Adding-Devices
I have added a Fibaro device FGS224 in my zwave network which is a double relay. It is the successor of fhs223  It has two individual loads controlled two switches S1 and S2. The states of both switches are sent to the gateway correctly when they are operated by the software. The state of S1 is also sent to the gateway when it is operated using the switch. But when I try to operate S2 using the actual switch its state is never updated. I changed the association to group 1 {"association": "remove", "node_id": 44, "group": 1, "target_node_id":1} and then {"association": "add", "node_id": 44, "group": 1, "target_node_id":1, "instance":1} After that the state of S2 is sent correctly but the state of S1 is not i.e it is reversed.
I have tried to add it in secure mode but nothing changes.

lucode, 2021-02-10:
Similar issue here with FGS224 with latest OpenZWave version: 0.9.1.  in HA (OZW 1.6.1789). But in my case I assume that the issue is caused by the faulty association/inclusion in encrypted mode.
[image]

In the image one can see that Instance 1 is secured Instance 2 not. With the other Fibaro switches both are secured (encrypted). -suppose this is a bug.

03397, 2021-02-10:
Have you tried in adding it in unsecure?

gxapplications, 2022-08-29:
Hello ! With last version, always the same problem, association is unsecure for me, have to use a poll to bypass...