OpenZWave/open-zwave #2593: Value updated/target value loop

eirikgund, 2021-03-05:
When manually dimming a Heatit Z-dim dimmer using the dimming wheel z-wave goes in to a constant loop. Where it seems to me like target value is never met.

> z2m:Zwave zwave node 32: changed: 38-1-9:Target Value:45 -> 45 +294ms
z2m:Zwave zwave node 32: changed: 38-1-5:Dimming Duration:0 -> 0 +1ms
OpenZWave Detail, Node032, Value Updated: old value=0, new value=0, type=int
OpenZWave Detail, Node032, Changes to this value are not verified
OpenZWave Detail, Node032, Expected reply and command class was received
OpenZWave Detail, Node032, Message transaction complete
OpenZWave Detail,
OpenZWave Detail, Node032, Removing current message
OpenZWave Detail, Node032, Notification: ValueRefreshed CC: COMMAND_CLASS_SWITCH_MULTILEVEL Instance: 1 Index: 9
OpenZWave Detail, Node032, Notification: ValueRefreshed CC: COMMAND_CLASS_SWITCH_MULTILEVEL Instance: 1 Index: 5
OpenZWave Info, Timer: delayed event
OpenZWave Info, Node032, Sending Get to Refresh Value after Target Check
OpenZWave Detail, Node032, Queuing (Send) SwitchMultilevelCmd_Get (Node=32): 0x01, 0x09, 0x00, 0x13, 0x20, 0x02, 0x26, 0x02, 0x25, 0x40, 0x86
OpenZWave Detail, Timer: waiting with timeout -1 ms
OpenZWave Detail,
OpenZWave Info, Node032, Sending (Send) message (Callback ID=0x40, Expected Reply=0x04) - SwitchMultilevelCmd_Get (Node=32): 0x01, 0x09, 0x00, 0x13, 0x20, 0x02, 0x26, 0x02, 0x25, 0x40, 0x86
OpenZWave Info, Node032, Encrypted Flag is 0
OpenZWave Detail, Node032, Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
OpenZWave Detail, Node032, ZW_SEND_DATA delivered to Z-Wave stack
OpenZWave Detail, Node032, Received: 0x01, 0x07, 0x00, 0x13, 0x40, 0x00, 0x00, 0x03, 0xa8
OpenZWave Detail, Node032, ZW_SEND_DATA Request with callback ID 0x40 received (expected 0x40)
OpenZWave Info, Node032, Request RTT 24 Average Request RTT 23
OpenZWave Detail, Node032, Expected callbackId was received
OpenZWave Detail, Node032, Received: 0x01, 0x0b, 0x00, 0x04, 0x00, 0x20, 0x05, 0x26, 0x03, 0x25, 0x2d, 0x00, 0xf8
OpenZWave Detail,
OpenZWave Info, Node032, Response RTT 39 Average Response RTT 38
OpenZWave Info, Node032, Received SwitchMultiLevel report: level=37
OpenZWave Detail, Node032, Value Updated: old value=45, new value=37, type=byte
OpenZWave Detail, Node032, Target Value is Set to 45
z2m:Zwave zwave node 32: changed: 38-1-9:Target Value:45 -> 45 +290ms
OpenZWave Info, Timer: adding event in 250 ms
z2m:Zwave zwave node 32: changed: 38-1-5:Dimming Duration:0 -> 0 +2ms
OpenZWave Detail, Node032, Value Updated: old value=45, new value=45, type=byte
OpenZWave Detail, Node032, Changes to this value are not verified
OpenZWave Detail, Timer: waiting with timeout 250 ms
OpenZWave Detail, Node032, Value Updated: old value=0, new value=0, type=int
OpenZWave Detail, Node032, Changes to this value are not verified
OpenZWave Detail, Node032, Expected reply and command class was received
OpenZWave Detail, Node032, Message transaction complete
OpenZWave Detail,
OpenZWave Detail, Node032, Removing current message
OpenZWave Detail, Node032, Notification: ValueRefreshed CC: COMMAND_CLASS_SWITCH_MULTILEVEL Instance: 1 Index: 9
OpenZWave Detail, Node032, Notification: ValueRefreshed CC: COMMAND_CLASS_SWITCH_MULTILEVEL Instance: 1 Index: 5
OpenZWave Info, Timer: delayed event
OpenZWave Info, Node032, Sending Get to Refresh Value after Target Check
OpenZWave Detail, Node032, Queuing (Send) SwitchMultilevelCmd_Get (Node=32): 0x01, 0x09, 0x00, 0x13, 0x20, 0x02, 0x26, 0x02, 0x25, 0x41, 0x87
OpenZWave Detail, Timer: waiting with timeout -1 ms
OpenZWave Detail,
OpenZWave Info, Node032, Sending (Send) message (Callback ID=0x41, Expected Reply=0x04) - SwitchMultilevelCmd_Get (Node=32): 0x01, 0x09, 0x00, 0x13, 0x20, 0x02, 0x26, 0x02, 0x25, 0x41, 0x87
OpenZWave Info, Node032, Encrypted Flag is 0
OpenZWave Detail, Node032, Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
OpenZWave Detail, Node032, ZW_SEND_DATA delivered to Z-Wave stack
OpenZWave Detail, Node032, Received: 0x01, 0x07, 0x00, 0x13, 0x41, 0x00, 0x00, 0x02, 0xa8
OpenZWave Detail, Node032, ZW_SEND_DATA Request with callback ID 0x41 received (expected 0x41)
OpenZWave Info, Node032, Request RTT 25 Average Request RTT 24
OpenZWave Detail, Node032, Expected callbackId was received
OpenZWave Detail, Node032, Received: 0x01, 0x0b, 0x00, 0x04, 0x00, 0x20, 0x05, 0x26, 0x03, 0x25, 0x2d, 0x00, 0xf8
z2m:Zwave zwave node 32: changed: 38-1-9:Target Value:45 -> 45 +291ms
z2m:Zwave zwave node 32: changed: 38-1-5:Dimming Duration:0 -> 0 +1ms
OpenZWave Detail,
OpenZWave Info, Node032, Response RTT 38 Average Response RTT 38
OpenZWave Info, Node032, Received SwitchMultiLevel report: level=37
OpenZWave Detail, Node032, Value Updated: old value=45, new value=37, type=byte
OpenZWave Detail, Node032, Target Value is Set to 45
OpenZWave Info, Timer: adding event in 250 ms
OpenZWave Detail, Timer: waiting with timeout 249 ms
OpenZWave Detail, Node032, Value Updated: old value=45, new value=45, type=byte
OpenZWave Detail, Node032, Changes to this value are not verified
[... 176 lines cut ...]
z2m:Zwave zwave node 32: changed: 38-1-9:Target Value:45 -> 45 +42ms
z2m:Zwave zwave node 32: changed: 38-1-5:Dimming Duration:0 -> 0 +2ms
OpenZWave Detail, Node032, Value Updated: old value=45, new value=37, type=byte
OpenZWave Detail, Node032, Target Value is Set to 45
OpenZWave Info, Timer: adding event in 250 ms
OpenZWave Detail, Node032, Value Updated: old value=45, new value=45, type=byte
OpenZWave Detail, Timer: waiting with timeout 206 ms
OpenZWave Detail, Node032, Changes to this value are not verified
OpenZWave Detail, Node032, Value Updated: old value=0, new value=0, type=int
OpenZWave Detail, Node032, Changes to this value are not verified
OpenZWave Detail, Node032, Expected reply and command class was received
OpenZWave Detail, Node032, Message transaction complete
OpenZWave Detail,
OpenZWave Detail, Node032, Removing current message
OpenZWave Detail, Node032, Notification: ValueRefreshed CC: COMMAND_CLASS_SWITCH_MULTILEVEL Instance: 1 Index: 9
OpenZWave Detail, Node032, Notification: ValueRefreshed CC: COMMAND_CLASS_SWITCH_MULTILEVEL Instance: 1 Index: 5
OpenZWave Info, Timer: delayed event
OpenZWave Info, Node032, Sending Get to Refresh Value after Target Check
OpenZWave Detail, Node032, Queuing (Send) SwitchMultilevelCmd_Get (Node=32): 0x01, 0x09, 0x00, 0x13, 0x20, 0x02, 0x26, 0x02, 0x25, 0x47, 0x81
OpenZWave Detail, Timer: waiting with timeout 41 ms
OpenZWave Detail,
OpenZWave Info, Node032, Sending (Send) message (Callback ID=0x47, Expected Reply=0x04) - SwitchMultilevelCmd_Get (Node=32): 0x01, 0x09, 0x00, 0x13, 0x20, 0x02, 0x26, 0x02, 0x25, 0x47, 0x81
OpenZWave Info, Node032, Encrypted Flag is 0
OpenZWave Detail, Node032, Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
OpenZWave Detail, Node032, ZW_SEND_DATA delivered to Z-Wave stack

rrozema, 2021-04-26:
The same thing happened for me with the MCOHome MH-DT411 Dimmer Switch. I fixed it for this device by setting it's option "12: Reporting interval for dimming level" to 0. This made the device's behavior better suited to openzwave's characteristics.
[image]
Does your device have a similar option?

rrozema, 2021-04-26:
I looked up the manual for your device. I think if you set option 9 to 0 for yours, the problem is gone:
Parameter 9, Parameter Size 2. Meter Report Interval.
The parameter defines the report interval for Power Metering.
VALUE DESCRIPTION
0 Disabled
30 - 32 767 Duration in seconds (30 - 32 767. Default is 60)

One note: I see the parameter size for this is option is 2. There is an error in CommandClass::ExtractValue() that can cause 2 and 4 byte values to be misread by openzwave (mostly affects negative values only), so you should be careful with the values shown for this option. 0 however should never be misinterpreted, so you should be fine here.

rrozema, 2021-04-26:
I have looked up the manual for your device, but didn't find a setting like on mine. What you could try however is set option 8 to 0. My guess is that should avoid sending intermediate updates, because the dimmer goes from its current value to the new value instantly.

Parameter 8, Parameter Size 1. Dimming Duration.
The parameter defines how long it takes to dim to the desired level.
Note! Only Multilevel Switch Command Class is affected by this parameter.
VALUE DESCRIPTION
0 Disabled
1 - 127 Duration in seconds (1 - 127 seconds. Default is 1)
128 - 255 Duration in minutes (1 - 127 minutes)