OpenZWave/open-zwave #2608: Error relating to serial port while compiling the code of openzwave

abir-wind, 2021-06-01:
hi , 
i'm facing this problem anyone can help , [](https://askubuntu.com/questions/1340583/error-relating-to-serial-port-while-compiling-the-code-of-openzwave )
thanks ,

igorlopez, 2021-08-06:
Exactly the same problem with my RAZBERRY2 (ZME*RAZ2).  
Have seen others saying it works when selecting different device but I have tried with both /dev/ttyAMA0, /dev/ttyS0 , /dev/serial0 and serial1.
Different error responses but none is working:  

    MinOZW /dev/ttyAMA0     
    ...    
    2021-08-06 13:23:57.887 Info,   Opening controller /dev/ttyAMA0  
    2021-08-06 13:23:57.887 Info, Starting DNSThread  
    2021-08-06 13:23:57.887 Info, Trying to open serial port /dev/ttyAMA0 (attempt 1)  
    2021-08-06 13:23:57.887 Info, Timer: thread starting  
    2021-08-06 13:23:57.887 Detail, Timer: waiting with timeout -1 ms  
    2021-08-06 13:23:57.887 Info, LookupTxT Checking mfs.db.openzwave.com  
    2021-08-06 13:23:57.887 Error, ERROR: Cannot open serial port /dev/ttyAMA0. Error code 13  
    2021-08-06 13:23:57.888 Error, ERROR: Failed to open serial port /dev/ttyAMA0  
    2021-08-06 13:23:57.888 Warning, WARNING: Failed to init the controller (attempt 0)  
  
    MinOZW /dev/serial0  
    ...  
    2021-08-06 13:30:25.993 Info,   Opening controller /dev/serial0  
    2021-08-06 13:30:25.993 Info, Starting DNSThread  
    2021-08-06 13:30:25.993 Info, Trying to open serial port /dev/serial0 (attempt 1)  
    2021-08-06 13:30:25.994 Info, mgr,     Added driver for controller /dev/serial0  
    2021-08-06 13:30:25.994 Info, LookupTxT Checking mfs.db.openzwave.com  
    2021-08-06 13:30:25.994 Error, ERROR: Cannot open serial port /dev/serial0. Error code 13  
    2021-08-06 13:30:25.994 Error, ERROR: Failed to open serial port /dev/serial0  
    2021-08-06 13:30:25.994 Warning, WARNING: Failed to init the controller (attempt 0)  
  
    MinOZW /dev/serial1  
    ...  
    2021-08-06 13:31:49.482 Info,   Opening controller /dev/serial1  
    2021-08-06 13:31:49.482 Info, Starting DNSThread  
    2021-08-06 13:31:49.482 Info, Trying to open serial port /dev/serial1 (attempt 1)  
    2021-08-06 13:31:49.482 Info, LookupTxT Checking mfs.db.openzwave.com  
    2021-08-06 13:31:49.482 Info, Timer: thread starting  
    2021-08-06 13:31:49.483 Detail, Timer: waiting with timeout -1 ms  
    2021-08-06 13:31:49.483 Error, ERROR: Failed to set serial port parameters  
    2021-08-06 13:31:49.483 Error, ERROR: Failed to open serial port /dev/serial1  
    2021-08-06 13:31:49.483 Warning, WARNING: Failed to init the controller (attempt 0)  
  
    MinOZW /dev/ttyS0  
    ...  
    2021-08-06 13:33:20.529 Info,   Opening controller /dev/ttyS0  
    2021-08-06 13:33:20.529 Info, Trying to open serial port /dev/ttyS0 (attempt 1)  
    2021-08-06 13:33:20.530 Info, Timer: thread starting  
    2021-08-06 13:33:20.530 Detail, Timer: waiting with timeout -1 ms  
    2021-08-06 13:33:20.530 Info, Starting DNSThread  
    2021-08-06 13:33:20.530 Error, ERROR: Failed to set serial port parameters  
    2021-08-06 13:33:20.530 Error, ERROR: Failed to open serial port /dev/ttyS0  
    2021-08-06 13:33:20.530 Warning, WARNING: Failed to init the controller (attempt 0)

igorlopez, 2021-08-07:
After finding Francois blog: [Home automation](http://fcerbell.github.io/IoTInfrastructure/) and following his instructions under the chapter: Razberry hat installation the Hat is at least found:  

    MinOZW /dev/ttyUSB20  
    ...  
    2021-08-07 10:23:13.852 Info,   Opening controller /dev/ttyUSB20  
    2021-08-07 10:23:13.853 Info, Trying to open serial port /dev/ttyUSB20 (attempt 1)  
    2021-08-07 10:23:13.853 Info, Starting DNSThread  
    2021-08-07 10:23:13.853 Info, Timer: thread starting  
    2021-08-07 10:23:13.853 Info, Serial port /dev/ttyUSB20 opened (attempt 1)  
    2021-08-07 10:23:13.853 Info, LookupTxT Checking mfs.db.openzwave.com  
    2021-08-07 10:23:13.853 Info, mgr,     Added driver for controller /dev/ttyUSB20  
    2021-08-07 10:23:13.853 Detail, Timer: waiting with timeout -1 ms  
    2021-08-07 10:23:13.854 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9  
    2021-08-07 10:23:13.856 Info, Lookup for mfs.db.openzwave.com returned 151  
    2021-08-07 10:23:15.996 Info, ManufacturerSpecificDB Initialized  
    2021-08-07 10:23:15.996 Detail, Notification: ManufacturerSpecificDB Ready  
    2021-08-07 10:23:15.996 Detail,   
    2021-08-07 10:23:15.996 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x15) - FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9  
    2021-08-07 10:23:15.996 Info, contrlr, Encrypted Flag is 0  
    2021-08-07 10:23:15.999 Detail, contrlr,   Received: 0x01, 0x10, 0x01, 0x15, 0x5a, 0x2d, 0x57, 0x61, 0x76, 0x65, 0x20, 0x36, 0x2e, 0x30, 0x32, 0x00, 0x01, 0x92  
    2021-08-07 10:23:15.999 Detail,   
    2021-08-07 10:23:15.999 Info, contrlr, Received reply to FUNC_ID_ZW_GET_VERSION:  
    2021-08-07 10:23:15.999 Info, contrlr,     Static Controller library, version Z-Wave 6.02  

So now I have to investigate how to attach my devices.