OpenZWave/open-zwave #2634: Where is the master branch to clone with git

sekatt, 2022-04-21:
Please answer me ... Help lol
Would like instead

sekatt, 2022-04-22:
> Please answer me ... Help lol
> Would like instead
> Tyvm
> It s ok 
> Bye

Likes

haydonryan, 2024-11-08:
I know this is really late, but when you git clone - you get the master branch. if you want to check out a specific version you can do something like this:

```
$git clone git@github.com:OpenZWave/open-zwave.git

$ git tag --list
Rel-1.0
V1.5
v1.2
v1.3
v1.4
v1.6
$ git checkout v1.6
Note: switching to 'v1.6'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 890f24b7 This is Version 1.6
```

Please close this issue, assuming this answers your quetsion