OpenZWave/open-zwave #2644: Web site with documentation is not accessible

ikhvostenkov, 2022-10-13:
http://www.openzwave.com site referenced in the README is not working. Did you move to the other host location? This would be helpful if this would be possible to access extended documentation.

BilalAzhar0, 2022-11-04:
Website is still not accessible....

blhoward2, 2022-11-16:
That’s because the project is all but dead after the original maintainer left. The website has been down for months. I’d suggest you give zwavejs a look as a replacement.

nechry, 2022-11-17:
@blhoward2 I am not the owner of the openzwave.org website and I have no admin access to the site. From my side I can not do anything to correct.

On the other hand, openzwave doesn't have maintainers either.

blhoward2, 2022-11-17:
Yes, I knew you didn’t own the domain. I thought you were maintaining the project, though, so I meant an update about the project altogether.

nechry, 2022-11-18:
Yes I have participated for a very long time, mainly in the quality control of configuration files. 
Without asking, I inherited the project, but I don't have the C++ skills required to follow the required changes.
So yes on paper, I'm the maintainer, but in fact it's something else...

I think openzwave will have been a great adventure that will even have helped to democratize Z-Wave to a greater mass of users.

Most of the home automation solutions are passed on Z-Wave JS all like me. Personally, I will invest my energies in Z-Wave JS.

If there are contributors interested and willing to continue openzwave, the door is open as in the past.