OpenZWave/open-zwave #2646: This project is no longer actively maintained

nechry, 2022-11-17:
I've kept this project alive for some time after I took ownership.
I have no time to try and maintain it, and it's wrong of me to pretend like I am maintaining it, I'm not a C++ developer.

I'm planning to set this repo as archived to signal that fact.
If anyone is interested in taking over ownership of it, feel free to reach out to me to chat.

It works as is, and will likely continue to do so for some time.

petergebruers, 2022-11-18:
Thank you for keeping it afloat for so long.

I have briefly worked on the C++ part and I merged plenty of config scripts.

If you want to get rid of the ownership, I can take over from you and Justin. But I would probably do the same thing: archive it, so people can still investigate and learn.

rzr, 2023-01-03:
May this project be transfered to @Abandonware org see :

https://purl.org/rzr/abandonware

If any interest let me know

chmeeedalf, 2025-02-04:
Having been burned by the latest zwave-js updates, I'd love to help contribute to this, and make it (or a wrapper for it) a viable alternative to zwave-js (drop-in replacement) for use with Home Assistant.